riscv-collab/riscv-gnu-toolchain #773: Bug in -fno-zero-initialized-in-bss?

alexthomas1, 2020-11-13:
Hello!

So I am trying to force the .bss section to be put into .data and I was using this flag; however, I don't think it works as intended here. 

The following code still creates a .bss section with the flag. 

` char arr[1024]; `

but 

`char arr[1024] = {0} `

does not. I thought that all global variables were initialized to 0 regardless. Is there a way to have the former code also be put in the .bss section? I also tried `-fcommon` and that did not work either. Thank you!

jim-wilson, 2020-11-14:
I believe -fno-zero-initialized-in-bss is working as intended.  "char arr[1024];" is uninitialized.  "char arr[1024] = { 0 };" is zero initialized.  So only the latter is affected by -fno-zero-initialized-in-bss.

Yes, the C standard says that uninialized variables get set to zero.  But that isn't the point here.  The point is that if the programmer initializes it to zero, then it doesn't go in bss.  This gives people a choice of whether variables go into data or bss depending on whether you choose to initialize it.  Maybe this doesn't make much sense anymore, but this option was invented a long time ago, and it did make sense then.

This isn't a RISC-V issue.  The option works the same for all targets.  If you think this should be changed, it would need to be discussed upstream.

alexthomas1, 2020-11-14:
Is there any option that you know of that can get me the intended results?

jim-wilson, 2020-11-15:
I think that there is no option that does exactly what you want.  You can explicitly initialize all variables.  Or you could try modifying gcc sources.  In the function bss_initializer_p in the file gcc/varasm.c there is the code
              || (flag_zero_initialized_in_bss
                  && initializer_zerop (DECL_INITIAL (decl)))));
you could change that to
              || flag_zero_initialized_in_bss));
and then it should work regardless of whether the variable is initialized or not.  Or maybe just change the function to always return false if you never want anything in bss.

There is some similar code farther down to handle thread local variables in bss, but a bare metal target is probably not using thread local variables.  You could try changing that one too for completeness.  Just look for the other use of flag_zero_initialized_in_bss.

TommyMurphyTM1234, 2022-09-27:
Questions about `-fno-zero-initialized-in-bss` and not an issue with the RISC-V GCC toolchain itself.

closed 2022-09-27 by TommyMurphyTM1234