riscv-collab/riscv-gnu-toolchain #774: Undecleared vector extension functions in branch rvv-intrinsic

lzwButton, 2020-11-16:
Hello.
In branch rvv-intrinsic, I am using vector extension to compile and found that some "vector widening integer reduction functions" are not declared.
For example, "vwredsum_vs_i16m8_i32m1" is not declared.

kito-cheng, 2020-11-17:
Confirmed, thanks for report that :)

lzwButton, 2020-11-17:
By the way, I also found that "mf2" related functions in "Vector Narrowing Fixed-Point Clip Functions" are not declared too. 
For example, "vnclip_wx_i32mf2" is not declared.

kito-cheng, 2020-11-17:
Fractional LMUL are not implement on GCC yet, so...that expected :P

lzwButton, 2020-11-17:
Got it! Thank you very much.

lzwButton, 2020-11-17:
I wonder when will all the LMUL be completed on GCC? Is there a date, like in two or three month.

kito-cheng, 2020-11-18:
Here is PR for fixing this issue: https://github.com/riscv/riscv-gcc/pull/214

For fractional LMUL...I can't guarantee any schedule here :P

kito-cheng, 2020-11-18:
But I would suggest you create another issue to track the fractional LMUL.

lzwButton, 2020-11-18:
Thank you! I will test it later and close the issue if it solved the problem.

lzwButton, 2020-11-18:
I tested and the problem solved. Therefore, the issue will be closed.

closed 2020-11-18 by lzwButton