riscv-collab/riscv-gnu-toolchain #775: Feature Request:  dump out --with-arch options the compiler was built with

MarcKarasek, 2020-11-16:
option for gcc to printout what arch extensions it was built with.  

for example built with --with-arch=rv64imafdc  some option to dump rv64imafdc to console. 

Something like --arch

jim-wilson, 2020-11-16:
gcc -v will show the configure options, including the --with-arch and --with-abi if any.

touch tmp.c; gcc -v tmp.c will show default options passed to cc1 and as which will show the default arch and abi options.

touch tmp.c; gcc -E -dM | grep __riscv will show the default predefined RISC-V macros.  This doesn't give arch directly, but will give xlen and all extensions.  It also give the abi by combining xlen and the float abi.

If you have an embedded elf compiler, then "touch tmp.c; gcc -S tmp.c; grep arch tmp.s" will give arch.

touch tmp.c; gcc -c tm-.c; readelf -h tmp.o will give float ABI.  You can get xlen from whether this is ELF32 or ELF64.  That gives you ABI.

MarcKarasek, 2020-11-16:
thanks for the info..  -v option is not very well documented in the--help

It says : Display the programs invoked by the compiler.

Marc

TommyMurphyTM1234, 2022-10-03:
Duplicate issue?

https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1101

TommyMurphyTM1234, 2022-11-11:
`-dumpspecs` also shows the default arch/abi:
```
./riscv32-unknown-linux-gnu-gcc -dumpspecs 
...
*multilib_defaults:
march=rv32imafdc mabi=ilp32d
...
```
On this basis I'm closing this issue.

closed 2022-11-11 by TommyMurphyTM1234