riscv-collab/riscv-gnu-toolchain #777: GDB cannot recognize vector instructions?

xicongye, 2020-11-18:
Hi,
I use 'rvv-intrinsic' branch to compile vector instructions, and run it on the spike, everything looks fine. But when I debug with gdb, I found gdb cannot show vector instructions correctly:
```
Dump of assembler code for function main:
   0x000000004000017e <+0>:	addi	sp,sp,-16
   0x0000000040000180 <+2>:	sd	s0,8(sp)
   0x0000000040000182 <+4>:	addi	s0,sp,16
=> 0x0000000040000184 <+6>:	li	t5,10
   0x0000000040000186 <+8>:	0x11f7357
   0x000000004000018a <+12>:	0xf6208457
   0x000000004000018e <+16>:	li	a5,0
   0x0000000040000190 <+18>:	mv	a0,a5
   0x0000000040000192 <+20>:	ld	s0,8(sp)
   0x0000000040000194 <+22>:	addi	sp,sp,16
   0x0000000040000196 <+24>:	ret
End of assembler dump.
```
Actually, the source code is :
```
int main(void)
{
  asm (
     "li t5, 10\n"
     "vsetvli t1, t5, e32, m2\n"
     "vqmacc.vv v8, v1, v2\n"
   );  
}
```
My steps to build toolchain are very simple:
```
../configure --with-arch=rv64gcv --with-abi=lp64d
```
So, what I'm doing wrong that causes this?

jim-wilson, 2020-11-18:
This is because binutils and gdb are built from different FSF release branches, and we put the vector instruction support on the binutils branch but not the gdb branch.  Since this is experimental work, we could fix it by building gdb from the binutils branch which has the vector support patches.  There is no guarantee that gdb on a binutils release branch is actually working but this is an experimental release so that is perhaps OK.

Another solution is to stop using a binutils release branch for binutils and a gdb release branch for gdb, and just use a binutils-gdb top of tree master branch, and use that for both the binutils and gdb builds.  We want to upstream this eventually anyways, and the FSF will only accept patches for the current top of tree.

Another solution is to copy all of the vector instruction support from the binutils branch to the gdb branch.  But then we have to remember to update it twice every time we modify it, because we have two copies of it on two different banches.  We need everything from the top of binutils down to the commit that marks the 2.35 binutils release copied over to the gdb tree.  And hope that it applies.  We probably only need the changes for the include and opcode dirs and can ignore the bfd/gas changes, as gdb in theory should not need those.

We will also have the same problem with other extensions, like the zfh extension, as we have assembler/disaassembler support for them on the binutils branch, but nothing on the gdb branch.

Nelson1225, 2020-11-19:
> Since this is experimental work, we could fix it by building gdb from the binutils branch which has the vector support patches. There is no guarantee that gdb on a binutils release branch is actually working but this is an experimental release so that is perhaps OK.

I will vote for this method, if users want to use some experimental features, like v-ext, zfh-ext or b-ext.  Although there is no guarantee that gdb on a binutils release branch is actually working, but at least the dis-assembler can work.

> Another solution is to stop using a binutils release branch for binutils and a gdb release branch for gdb, and just use a binutils-gdb top of tree master branch, and use that for both the binutils [...]

Exclude the above special needs, use the FSF binutils + FSF gdb should be the ideally combination.  But we may still get some unknown errors since the master branches are also under development.  Perhaps the latest FSF released branches are more stable to use?

> Another solution is to copy all of the vector instruction support from the binutils branch to the gdb branch. But then we have to remember to update it twice every time we modify it, because we have two copies of it on two different branches. We need everything from the top of binutils down to the commit that marks the 2.35 binutils release copied over to the gdb tree. And hope that it applies. We probably only need the changes for the include and opcode dirs and can ignore the bfd/gas changes, as gdb in theory should not need those.

I hope we will not have such a demand in the future...

> We will also have the same problem with other extensions, like the zfh extension, as we have assembler/disaassembler support for them on the binutils branch, but nothing on the gdb branch.

Yeah, and there may be more in the future, once we have more extensions/features are under development...

TommyMurphyTM1234, 2022-09-29:
No significant update/activity in almost two years.
And, nowadays, the correct place to look for ongoing implementation/enhancement of V extension v1.0 support is the `rvv-next` branch (https://github.com/riscv-collab/riscv-gnu-toolchain/tree/rvv-next).

TommyMurphyTM1234, 2022-10-25:
I tried this with the latest `rvv-next` branch and the issue persists:
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain -b rvv-next rvv
cd rvv
git log | head -1
commit da01ba455ce3802ffa84fdca3a089079996dbfc3
./configure --prefix=`pwd`/installed-tools --with-arch=rv64gcv
date > build.log && make 2>&1 | tee -a build.log && date >> build.log
cd installed-tools/bin
cat > test.c
int main(void)
{
  asm (
     "li t5, 10\n"
     "vsetvli t1, t5, e32, m2\n"
   );  
}
Ctrl-C
./riscv64-unknown-elf-gcc -g test.c
./riscv64-unknown-elf-gdb a.out
(gdb) disassemble main
Dump of assembler code for function main:
   0x0000000000010490 <+0>:	addi	sp,sp,-16
   0x0000000000010492 <+2>:	sd	s0,8(sp)
   0x0000000000010494 <+4>:	addi	s0,sp,16
   0x0000000000010496 <+6>:	li	t5,10
   0x0000000000010498 <+8>:	0x11f7357
   0x000000000001049c <+12>:	li	a5,0
   0x000000000001049e <+14>:	mv	a0,a5
   0x00000000000104a0 <+16>:	ld	s0,8(sp)
   0x00000000000104a2 <+18>:	addi	sp,sp,16
   0x00000000000104a4 <+20>:	ret
End of assembler dump.
(gdb) 
```
Note that the `vqmacc.vv v8, v1, v2` instruction is not valid in the latest RVV implementation so I had to omit it.

cmuellner, 2022-11-11:
Vector support is implemented and available upstream for Binutils/GDB (same repo).
So nothing is missing, but the wrong (extremely outdated) sources were used to build GDB.

@TommyMurphyTM1234: can you please test again and use the same sources for GDB as for binutils.

Background:
In the branch that you have referenced (rvv-next) the GDB is multiple years old.
We have fixed that recently on master (https://github.com/riscv-collab/riscv-gnu-toolchain/pull/1132), but the rvv-next branch is too old and does not have this change.

TommyMurphyTM1234, 2022-11-11:
> Vector support is implemented and available upstream for Binutils/GDB (same repo). So nothing is missing, but the wrong (extremely outdated) sources were used to build GDB.
>
> @TommyMurphyTM1234: can [...]

This is very confusing.
There are no clear instructions on how to build the latest/best RVV enabled toolchain.
I thought it was too checkout the `rvv-next` branch of `riscv-gnu-toolchain` (recursively) and then the `riscv-gcc-rvv-next` branch of `riscv-gcv`. 
Are you saying that that is NOT the correct way?
If so what is the correct way?

cmuellner, 2022-11-11:
Yes, this is the correct way.
But in order to get GDB working as well, you will have to get more recent GDB source.

cmuellner, 2022-11-11:
Probably you can also simply cherry-pick `b2f50de3b766c34ac166688e2586f9a2e974dbce` onto the `rvv-next` branch (untested).

TommyMurphyTM1234, 2022-11-11:
Any ideas?
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain issue-777 -b rvv-next --recursive
cd issue-777
git cherry-pick b2f50de3b766c34ac166688e2586f9a2e974dbce
Removing riscv-gdb
Adding as riscv-binutils~HEAD instead
Auto-merging configure.ac
Auto-merging configure
Auto-merging .gitmodules
error: could not apply b2f50de... gdb: Bump GDB version to 12.1 (#1132)
hint: after resolving the conflicts, mark the corrected paths
hint: with 'git add <paths>' or 'git rm <paths>'
hint: and commit the result with 'git commit'
```

cmuellner, 2022-11-11:
I think the `--recursive` is the issue here.
You first download the gdb submodule and then patch it away, which will cause these `auto-merging` and this will fail.

Try this:
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain issue-777 -b rvv-next
cd issue-777
git cherry-pick b2f50de3b766c34ac166688e2586f9a2e974dbce
git submodule update --init --recursive
##Fixup GCC sources
##Configure
##...
```

TommyMurphyTM1234, 2022-11-11:
Thanks - will try that now...

TommyMurphyTM1234, 2022-11-12:
Hi @cmuellner - thanks - that worked:
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain issue-777 -b rvv-next
cd issue-777/
git cherry-pick b2f50de3b766c34ac166688e2586f9a2e974dbce
git submodule update --init --recursive
cd riscv-gcc/
git checkout riscv-gcc-rvv-next
git pull
cd ..
./configure --prefix=`pwd`/installed-tools --with-arch=rv64gcv
make
cd installed-tools/bin/
cat > test.c
int main(void)
{
  asm (
     "li t5, 10\n"
     "vsetvli t1, t5, e32, m2\n"
   );  
}
Ctrl-C
./riscv64-unknown-elf-gcc -g test.c
./riscv64-unknown-elf-gdb a.out
(gdb) disassemble main
Dump of assembler code for function main:
   0x000000000001019c <+0>:	addi	sp,sp,-16
   0x000000000001019e <+2>:	sd	s0,8(sp)
   0x00000000000101a0 <+4>:	addi	s0,sp,16
   0x00000000000101a2 <+6>:	li	t5,10
   0x00000000000101a4 <+8>:	vsetvli	t1,t5,e32,m2,tu,mu
   0x00000000000101a8 <+12>:	li	a5,0
   0x00000000000101aa <+14>:	mv	a0,a5
   0x00000000000101ac <+16>:	ld	s0,8(sp)
   0x00000000000101ae <+18>:	addi	sp,sp,16
   0x00000000000101b0 <+20>:	ret
End of assembler dump.
```

closed 2022-11-12 by TommyMurphyTM1234

TommyMurphyTM1234, 2022-11-12:
Fixed by https://github.com/riscv-collab/riscv-gnu-toolchain/pull/1158