riscv-collab/riscv-gnu-toolchain #779: RVV Integer Scalar Moves cause internal compiler error on RV32 targets

MarkHillHuawei, 2020-11-20:
Internal compiler error occurs when trying to compile this code fragment  when using the rvv-intrinsics branch:

 ```
#include <riscv_vector.h>
int32_t x_to_s(int32_t i)
{
    vint32m8_t v;
    v    = vmv_s_x_i32m8(v, i);
    return vmv_x_s_i32m8_i32(v);
}
``` 

riscv32-unknown-elf-gcc -march=rv32gcv  -c test.c -S -o-

```
test.c: In function 'x_to_s':
test.c:11:1: error: unrecognizable insn:
   11 | }
      | ^
(insn 52 51 53 2 (parallel [
            (set (reg:VNx32SI 135 [ _10 ])
                (unspec:VNx32SI [
                        (vec_merge:VNx32SI (vec_duplicate:VNx32SI (reg:SI 176))
                            (reg:VNx32SI 175)
                            (const_int 1 [0x1]))
                        (reg:SI 66 vl)
                    ] UNSPEC_USEVL))
            (use (reg:VNx32SI 67 vtype))
        ]) "/data/toolchains/riscv/rv32gc/10.1.0-rvv-intrinsic/lib/gcc/riscv32-unknown-elf/10.1.0/include/riscv_vector.h":665:1 -1
     (nil))
during RTL pass: vregs
test.c:11:1: internal compiler error: in extract_insn, at recog.c:2294

```
Code compiles successfully if target is an RV64 machine:

riscv32-unknown-elf-gcc -march=rv64gcv -mabi=lp64 test.c -c -S -o-

kito-cheng, 2020-11-23:
Confirmed, thanks for report this issue, we were focusing on the RV64GCV support, on RV32GCV side we don't have too much strong verification, so I am really appreciate your bug report :)

kito-cheng, 2020-11-24:
@MarkHillHuawei Could you try this https://github.com/riscv/riscv-gcc/pull/218 ?

MarkHillHuawei, 2020-11-24:
Thanks Kito-Cheng, I have applied the patch, rebuilt the rv32 toolchain and can confirm the correct code is now generated.

Syndace, 2021-02-16:
Has this patch made it to the rvv-intrinsics branch yet? I'm still encountering the internal compiler error reported here on the most recent version of that branch. Or should I use a different branch for up-to-date intrinsics support?

jim-wilson, 2021-02-16:
The riscv-gnu-toolchain branch is correct, but the submodules may not be up to date.  Maintaining submodules is a pain, as you need two pull requests, one for riscv-gcc and then one for riscv-gnu-toolchain. We don't always bother with the second one.  Checking, it looks like riscv-gnu-toolchain's riscv-gcc is 6 commits behind the head of the branch and one is the patch you want.  You can just cd into riscv-gcc and fetch and pull the top of the riscv-gcc-10.1-rvv-dev branch.

lazyparser, 2021-03-03:
Seems the issue got fixed and merged. Can this issue be closed safely?

Syndace, 2021-03-03:
Yes, a fix is available, but that fix has not yet landed in this repository. I don't consider this "safe to close" until the submodule version has been bumped.

TommyMurphyTM1234, 2022-10-09:
Tested with the latest tools.

Branch: `rvv-next`
Commit: da01ba455ce3802ffa84fdca3a089079996dbfc3

Tools built as follows:
```
git clone https://github.com/riscv/riscv-gnu-toolchain rv32-rvv-next -b rvv-next
cd rvv-next
./configure --prefix=`pwd`/installed-tools --with-arch=rv32gcv
d

lhtin, 2022-10-10:
@TommyMurphyTM1234 the error means you forget the `dest` argument, the `vmv_s_x_i32m8` intrinisc prototype like this:

```c
vint32m8_t vmv_s_x_i32m8 (vint32m8_t dest, int32_t src, size_t vl);
```

TommyMurphyTM1234, 2022-10-10:
> ```c
> vint32m8_t vmv_s_x_i32m8 (vint32m8_t dest, int32_t src, size_t vl);
> ```

Thanks @lhtin - how can one see what the function prototypes are?

Anyway, I changed the test code to the following:
```
#include <riscv_vector.h>
int32_t x_to_s(int32_t i)
{
   vint32m8_t v;
   size_t vl;
   v    = vmv_s_x_i32m8(v, i, vl);
   return vmv_x_s_i32m8_i32(v);
}
```
and I get this:
```tommy@pelican:~/Downloads/riscv-tools/rv32-rvv-next/installed-tools/bin$ ./riscv32-unknown-elf-gcc -march=rv32gcv -c test.c -S -o-
	.file	"test.c"
	.option nopic
	.attribute arch, "rv32i2p0_m2p0_a2p0_f2p0_d2p0_c2p0_v1p0_zve32f1p0_zve32x1p0_zve64d1p0_zve64f1p0_zve64x1p0_zvl128b1p0_zvl32b1p0_zvl64b1p0"
	.attribute unaligned_access, 0
	.attribute stack_align, 16
	.text
	.align	1
	.globl	x_to_s
	.type	x_to_s, @function
x_to_s:
	addi	sp,sp,-48
	sw	s0,44(sp)
	addi	s0,sp,48
	csrr	t0,vlenb
	slli	t1,t0,3
	sub	sp,sp,t1
	csrr	a5,vlenb
	neg	a5,a5
	slli	a5,a5,3
	addi	a5,a5,-20
	addi	a5,a5,-16
	add	a5,a5,s0
	sw	a0,0(a5)
	csrr	a5,vlenb
	neg	a5,a5
	slli	a5,a5,3
	addi	a5,a5,-16
	add	a4,a5,s0
	csrr	a5,vlenb
	neg	a5,a5
	slli	a5,a5,3
	addi	a5,a5,-16
	add	a2,a5,s0
	csrr	a5,vlenb
	neg	a5,a5
	slli	a5,a5,3
	addi	a5,a5,-20
	addi	a5,a5,-16
	add	a3,a5,s0
	csrr	a5,vlenb
	neg	a5,a5
	slli	a5,a5,3
	addi	a5,a5,-4
	addi	a5,a5,-16
	add	a5,a5,s0
	vl8re32.v	v24,0(a2)
	lw	a3,0(a3)
	lw	a5,0(a5)
	vsetvli	zero,a5,e32,m8,tu,mu
	vmv.s.x	v24,a3
	vs8r.v	v24,0(a4)
	csrr	a4,vlenb
	neg	a4,a4
	slli	a4,a4,3
	addi	a4,a4,-16
	add	a4,a4,s0
	vl8re32.v	v24,0(a4)
	vsetvli	zero,zero,e32,m8,ta,mu
	vsetivli	zero,0,e32,m8,ta,mu
	vmv.x.s	a5,v24
	mv	a0,a5
	csrr	t0,vlenb
	slli	t1,t0,3
	add	sp,sp,t1
	lw	s0,44(sp)
	addi	sp,sp,48
	jr	ra
	.size	x_to_s, .-x_to_s
	.ident	"GCC: () 12.0.1 20220505 (prerelease)
```
So, it seems safe to conclude that the original issue is resolved in the latest `rvv-next` branch.

closed 2022-10-10 by TommyMurphyTM1234

lhtin, 2022-10-10:
> how can one see what the function prototypes are?

You can find the official documents about these RVV intrinsics from [there](https://github.com/riscv-non-isa/rvv-intrinsic-doc) which is for RVV 1.0. The intrinsic `vmv_s_x_i32m8` locate [there](https://github.com/riscv-non-isa/rvv-intrinsic-doc/blob/master/intrinsic_funcs/09_vector_permutation_functions.md#integer-and-floating-point-scalar-move-functions)

TommyMurphyTM1234, 2022-10-10:
> You can find the official documents about these RVV intrinsics from [there](https://github.com/riscv-non-isa/rvv-intrinsic-doc) which is for RVV 1.0. The intrinsic `vmv_s_x_i32m8` locate [...]

Thanks @lhtin.