riscv-collab/riscv-gnu-toolchain #780: missing operand for /usr/bin/install when running make linux

MartinGeisse, 2020-11-22:
Host system: Fedora

Commands:
./configure --prefix=/home/martin.geisse/git-repos/riscv-linux-toolchain/install-path --with-arch=rv32ima --with-abi=ilp32
make linux

End of output:
```
/usr/bin/gawk -v buildlist="iconvprogs iconvdata ldconfig lddlibc4 libmemusage libSegFault libpcprofile librpcsvc locale_programs memusagestat nonlib nscd extramodules libnldbl libsupport testsuite" -f scripts/gen-libc-modules.awk /home/martin.geisse/git-repos/riscv-linux-toolchain/build-glibc-linux-rv32ima-ilp32/soversions.i > /home/martin.geisse/git-repos/riscv-linux-toolchain/build-glibc-linux-rv32ima-ilp32/libc-modules.T
/bin/sh scripts/move-if-change /home/martin.geisse/git-repos/riscv-linux-toolchain/build-glibc-linux-rv32ima-ilp32/libc-modules.T /home/martin.geisse/git-repos/riscv-linux-toolchain/build-glibc-linux-rv32ima-ilp32/libc-modules.h
touch /home/martin.geisse/git-repos/riscv-linux-toolchain/build-glibc-linux-rv32ima-ilp32/libc-modules.stmp
for dir in /home/martin.geisse/git-repos/riscv-linux-toolchain/build-glibc-linux-rv32ima-ilp32 sysdeps/unix/sysv/linux/riscv/rv32/ sysdeps/unix/sysv/linux/riscv sysdeps/riscv/nptl sysdeps/unix/sysv/linux/generic/wordsize-32 sysdeps/unix/sysv/linux/generic sysdeps/unix/sysv/linux/riscv/rv32 sysdeps/unix/sysv/linux sysdeps/nptl sysdeps/pthread sysdeps/gnu sysdeps/unix/inet sysdeps/unix/sysv sysdeps/unix sysdeps/posix sysdeps/riscv/rv32/ sysdeps/riscv/rv32 sysdeps/riscv/nofpu sysdeps/ieee754/soft-fp sysdeps/riscv sysdeps/init_array sysdeps/ieee754/ldbl-128 sysdeps/ieee754/dbl-64 sysdeps/ieee754/flt-32 sysdeps/wordsize-32 sysdeps/ieee754 sysdeps/generic; do \
  test -f $dir/syscalls.list && \
  { sysdirs='sysdeps/unix/sysv/linux/riscv/rv32/ sysdeps/unix/sysv/linux/riscv sysdeps/riscv/nptl sysdeps/unix/sysv/linux/generic/wordsize-32 sysdeps/unix/sysv/linux/generic sysdeps/unix/sysv/linux/riscv/rv32 sysdeps/unix/sysv/linux sysdeps/nptl sysdeps/pthread sysdeps/gnu sysdeps/unix/inet sysdeps/unix/sysv sysdeps/unix sysdeps/posix sysdeps/riscv/rv32/ sysdeps/riscv/rv32 sysdeps/riscv/nofpu sysdeps/ieee754/soft-fp sysdeps/riscv sysdeps/init_array sysdeps/ieee754/ldbl-128 sysdeps/ieee754/dbl-64 sysdeps/ieee754/flt-32 sysdeps/wordsize-32 sysdeps/ieee754 sysdeps/generic' \
    asm_CPP='riscv32-unknown-linux-gnu-gcc  -c     -Iinclude   -I/home/martin.geisse/git-repos/riscv-linux-toolchain/build-glibc-linux-rv32ima-ilp32  -Isysdeps/unix/sysv/linux/riscv/rv32/  -Isysdeps/unix/sysv/linux/riscv  -Isysdeps/riscv/nptl  -Isysdeps/unix/sysv/linux/generic/wordsize-32  -Isysdeps/unix/sysv/linux/generic  -Isysdeps/unix/sysv/linux/riscv/rv32  -Isysdeps/unix/sysv/linux/include -Isysdeps/unix/sysv/linux  -Isysdeps/nptl  -Isysdeps/pthread  -Isysdeps/gnu  -Isysdeps/unix/inet  -Isysdeps/unix/sysv  -Isysdeps/unix  -Isysdeps/posix  -Isysdeps/riscv/rv32/  -Isysdeps/riscv/rv32  -Isysdeps/riscv/nofpu  -Isysdeps/ieee754/soft-fp  -Isysdeps/riscv  -Isysdeps/init_array  -Isysdeps/ieee754/ldbl-128  -Isysdeps/ieee754/dbl-64  -Isysdeps/ieee754/flt-32  -Isysdeps/wordsize-32  -Isysdeps/ieee754  -Isysdeps/generic   -Ilibio -I. -nostdinc -isystem /home/martin.geisse/git-repos/riscv-linux-toolchain/install-path/lib/gcc/riscv32-unknown-linux-gnu/10.2.0/include -isystem /home/martin.geisse/git-repos/riscv-linux-toolchain/install-path/lib/gcc/riscv32-unknown-linux-gnu/10.2.0/include-fixed -isystem /home/martin.geisse/git-repos/riscv-linux-toolchain/linux-headers/include  -D_LIBC_REENTRANT -include /home/martin.geisse/git-repos/riscv-linux-toolchain/build-glibc-linux-rv32ima-ilp32/libc-modules.h -DMODULE_NAME=libc -include include/libc-symbols.h       -DTOP_NAMESPACE=glibc -DASSEMBLER  -g -Werror=undef -Wa,--noexecstack   -E -x assembler-with-cpp' \
    /bin/sh sysdeps/unix/make-syscalls.sh $dir || exit 1; }; \
  test $dir = sysdeps/unix && break; \
done > /home/martin.geisse/git-repos/riscv-linux-toolchain/build-glibc-linux-rv32ima-ilp32/sysd-syscallsT
mv -f /home/martin.geisse/git-repos/riscv-linux-toolchain/build-glibc-linux-rv32ima-ilp32/sysd-syscallsT /home/martin.geisse/git-repos/riscv-linux-toolchain/build-glibc-linux-rv32ima-ilp32/sysd-syscalls
rm -f /home/martin.geisse/git-repos/riscv-linux-toolchain/build-glibc-linux-rv32ima-ilp32/stamp.o; > /home/martin.geisse/git-repos/riscv-linux-toolchain/build-glibc-linux-rv32ima-ilp32/stamp.o
rm -f /home/martin.geisse/git-repos/riscv-linux-toolchain/build-glibc-linux-rv32ima-ilp32/stamp.os; > /home/martin.geisse/git-repos/riscv-linux-toolchain/build-glibc-linux-rv32ima-ilp32/stamp.os
rm -f /home/martin.geisse/git-repos/riscv-linux-toolchain/build-glibc-linux-rv32ima-ilp32/stamp.oS; > /home/martin.geisse/git-repos/riscv-linux-toolchain/build-glibc-linux-rv32ima-ilp32/stamp.oS
make  subdir=csu -C csu ..=../ subdir_lib
make[3]: Verzeichnis „/home/martin.geisse/git-repos/riscv-linux-toolchain/riscv-glibc/csu“ wird betreten
/bin/sh ../scripts/gen-libc-abis \
	 riscv-unknown-linux-gnu \
	 < ../libc-abis > /home/martin.geisse/git-repos/riscv-linux-toolchain/build-glibc-linux-rv32ima-ilp32/libc-abis.hT
/bin/sh ../scripts/move-if-change /home/martin.geisse/git-repos/riscv-linux-toolchain/build-glibc-linux-rv32ima-ilp32/libc-abis.hT /home/martin.geisse/git-repos/riscv-linux-toolchain/build-glibc-linux-rv32ima-ilp32/libc-abis.h
touch /home/martin.geisse/git-repos/riscv-linux-toolchain/build-glibc-linux-rv32ima-ilp32/libc-abis.stamp
/usr/bin/install -c -m 644  /home/martin.geisse/git-repos/riscv-linux-toolchain/build-glibc-linux-rv32ima-ilp32/cstdlibT
/usr/bin/install: Fehlender Zieldatei‐Operand hinter '/home/martin.geisse/git-repos/riscv-linux-toolchain/build-glibc-linux-rv32ima-ilp32/cstdlibT'
„/usr/bin/install --help“ liefert weitere Informationen.
make[3]: *** [../Makerules:132: /home/martin.geisse/git-repos/riscv-linux-toolchain/build-glibc-linux-rv32ima-ilp32/cstdlib] Fehler 1
make[3]: Verzeichnis „/home/martin.geisse/git-repos/riscv-linux-toolchain/riscv-glibc/csu“ wird verlassen
make[2]: *** [Makefile:258: csu/subdir_lib] Fehler 2
make[2]: Verzeichnis „/home/martin.geisse/git-repos/riscv-linux-toolchain/riscv-glibc“ wird verlassen
make[1]: *** [Makefile:9: all] Fehler 2
make[1]: Verzeichnis „/home/martin.geisse/git-repos/riscv-linux-toolchain/build-glibc-linux-rv32ima-ilp32“ wird verlassen
make: *** [Makefile:289: stamps/build-glibc-linux-rv32ima-ilp32] Fehler 2
```

The error message seems to be "Fehlender Zieldatei‐Operand hinter ..." (missing target file operand after ...) -- note that there is a suspicious double blank *before* that operand.

jim-wilson, 2020-11-22:
This is a build failure I haven't seen before.  A web search shows a few people have reported similar problems, but no solutions were reported.

In the glibc Makerules file, I see some rules to make a copy of the cstdlib C++ header file to work around a gcc-6 problem.  But this stuff is only used if CXX is set, and CXX should not be set here.  In my glibc build dir, in the config.make file, I have
CXX =
We build only C in the first stage gcc build, and then use it to build glibc, and then build C and C++ in the second stage gcc build. 
 So there should be no C++ compiler when building glibc.  The makefile rule gets the name of the cstdlib file from configure.  So that should also be in config.make.  In my build I have
c++-cstdlib-header =
because we haven't built libstdc++ yet which is done in the second stage gcc build.

So my theory is that you have a partially installed C++ compiler in the $prefix dir.  It has a g++ binary, but not a cstdlib header file.  So when make gets to this point in Makefules, because CXX is set it tries to copy the cstdlib header file, but since the header file does not exist you get an install command with only one file name as an argument.

The solution then would be to rm -rf your broken install, or else use a different $prefix if you can't delete it.

The README file does mention that trying to build overtop an existing install tree can fail.  it is recommend that you alway use an empty $prefix before starting a build.  Either using a unique prefix for every build, or else using rm -rf to remove the old one.

MartinGeisse, 2020-11-23:
A partial install cannot be the cause of the problem because I DO rm -rf the install path before building.

However, I noticed that another version of the toolchain in another path was still in the PATH. After I removed that, the toolchain builds without errors. 

Suggestions:
- the README should say that having a different version of the toolchain in the PATH is not okay, or
- the build scripts should stay within their folders and not pick up random tools from somewhere else just because they are in the PATH (platform-provided tools like the host GCC are obviously an exception to this)
- the build scripts should check if any of these error conditions are met, and print a useful message instead of running into follow-up problems later

TommyMurphyTM1234, 2022-09-29:
No update/activity or PR in almost two years. The README can't cover every possibility and users should know enough to do a toolchain build in a "clean" environment with possibly incompatible tools accessible.

closed 2022-09-29 by TommyMurphyTM1234