riscv-collab/riscv-gnu-toolchain #781: compile c++ program error

Dream-math, 2020-11-23:
My test.cpp is like this:

```
#include <iostream>
#include <mutex>

typedef std::once_flag once_flag;

int main(){
  std::cout<<"Hello RISCV!"<<std::endl;
  return 0;
}
```
I compile with
```
riscv64-unknown-elf-g++ test.cpp
```
the error message is
```
test.cpp:4:14: error: 'once_flag' in namespace 'std' does not name a type
    4 | typedef std::once_flag once_flag;
      |              ^~~~~~~~~
test.cpp:3:1: note: 'std::once_flag' is defined in header '<mutex>'; did you forget to '#include <mutex>'?
    2 | #include <mutex>
  +++ |+#include <mutex>
    3 | 
```
why this happens, please help me, thanks a lot. By the way, the hello world test is right.

jim-wilson, 2020-11-23:
An embedded elf compiler doesn't support threads.  Threads requires some sort of OS support to manage the threads, and an embedded elf compiler assumes no OS.

A riscv64-unknown-linux-gnu-g++ compiler does support threads, and can compile this testcase.

Dream-math, 2020-11-23:
Thank you very much, I know nothing about embedded things. I will learn more about it.

closed 2020-11-23 by Dream-math