riscv-collab/riscv-gnu-toolchain #782: How to Add custom instructions to riscv-elf-toolchain

rahulraveendran15-coder, 2020-11-23:
Hi,

I want to implement some custom instructions on riscv-architecture to speed up my process.
How can I add the custom instruction to the tool chain, so that on compilation it can get new instructions. Is it possible to add custom instructions to riscv-elf-tool chain? I am not finding any respective documents/reference for this

jim-wilson, 2020-11-23:
You can find gcc info at gcc.gnu.org.  There are some tutorials there under the "pointers" link but I'm not sure how good they are.

I've given brief overviews of the process in various places.  Here is one for instance
    https://groups.google.com/a/groups.riscv.org/g/sw-dev/c/sL_OHXYj3LY/m/Gsm6sBc9BQAJ

TommyMurphyTM1234, 2022-11-10:
This is a list of all issues logged that I can find that relate to adding new instructions to the toolchain:

* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1095
* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1055
* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/837
* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/782
* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1003
* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/795
* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/660
* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/543
* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/341
* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/115

At this stage in the evolution of RISC-V,the majority of instruction addition will be done under the auspices of the relevant groups working on standardising, prototyping, and implementing/enhancing RISC-V extensions and their corresponding instructions. As such, there should be less of a demand for others to do so.

However, where somebody still wants to implement a non stadard custom extension/instruction here are some useful resources:

* GCC internals documentation: https://gcc.gnu.org/onlinedocs/gccint/
* https://hsandid.github.io/posts/risc-v-custom-instruction/
* https://ossg.bcs.org/blog/2020/10/29/adding-an-instruction-to-the-gnu-assembler/
* https://embdev.net/topic/228773
* https://groups.google.com/u/0/a/groups.riscv.org/g/sw-dev/c/sL_OHXYj3LY/m/NOm5Fwa3BAAJ
* https://forums.sifive.com/t/adding-custom-instruction-to-riscv-toolchain/4044/2

closed 2022-11-10 by TommyMurphyTM1234