riscv-collab/riscv-gnu-toolchain #784: relocation truncated to fit: R_RISCV_HI20 against `.LC0'

suhasskrishanmurthy, 2020-11-25:
Hi,
   I am trying to compile whetstone code(got this code from netlib.org ) with arch rv64imafd abi=lp64d. compiling in riscv-tests repo.
compilation cmd:
riscv64-unknown-elf-gcc -I./../env -I./common  -I./whetstone -DPREALLOCATE=1 -mcmodel=medlow -g -march=rv64imafd -mabi=lp64d -std=gnu99 -O0 -ffast-math -fno-common -fno-builtin-printf -fpie -o whetstone.riscv ./whetstone/whetstone.c ./common/syscalls.c ./math_src/mtherr.c ./math_src/isnan.c ./math_src/const.c ./math_src/pow.c ./math_src/powi.c ./math_src/log.c ./math_src/sqrt.c ./math_src/acosh.c ./math_src/polevl.c ./math_src/floor.c ./common/crt.S -static -nostdlib -nostartfiles  -lm -lgcc -T ./common/test.ld

It shows error like this:
/home/riscv/riscv_toolchain/riscv-tools/lib/gcc/riscv64-unknown-elf/7.2.0/../../../../riscv64-unknown-elf/lib/libm.a(lib_a-s_atan.o): In function `.L4':
s_atan.c:(.text+0x46): relocation truncated to fit: R_RISCV_HI20 against `.LC0'
/home/riscv/riscv_toolchain/riscv-tools/lib/gcc/riscv64-unknown-elf/7.2.0/../../../../riscv64-unknown-elf/bin/ld: final link failed: Symbol needs debug section which does not exist
collect2: error: ld returned 1 exit status
Makefile:86: recipe for target 'whetstone.riscv' failed

I saw some previous post, it mentioned about keeping mcmodel with which toolchain is compiled. So I changed to that, but still same error.

Any suggestions on solving this issue would be helpful.

jim-wilson, 2020-11-25:
gcc-7.2.0 is old obsolete and known to be broken.  It should never be used for anything.

The benchmarks/common/test.ld linker script puts code at 0x80000000.  This is where rocket-chip puts DRAM.  Code loaded at 0x80000000 can only work if compiled -mcmodel=medany.  You specified -mcmodel=medlow which won't work.  Also, if you are linking with any precompiled libraries, like the C library, then the toolchain must have been configured --with-cmodel=medany or else the C library will be medlow code and won't work.

Chakri-myneni, 2020-12-01:
I was using gcc-7.2.0 faced issue similar to that of @suhasskrishanmurthy. 
I updated the gcc to latest version as @jim-wilson said and it's working fine for me now.

suhasskrishanmurthy, 2020-12-01:
@jim-wilson, thanks your suggestion worked for me. I am using 9.2.0 now

closed 2020-12-01 by suhasskrishanmurthy