riscv-collab/riscv-gnu-toolchain #785: Build and compilation commands usage

rahulraveendran15-coder, 2020-11-25:
Hi,
I have a doubt on build and compilation command usage.

Consider I am building the tool chain with "./configure --prefix=/opt/riscv --with-arch=rv32imc --with-abi=ilp32"

After building the tool chain with above command do I need to provide "--march=rv32imc" along with my compilation command like below? What is the use or benefit of adding --march along with the compilation command?

riscv32-unknown-elf-gcc --march=rv32imc --mabi=ilp32 -o hello_world hello_world.c

Same concern when building the tool chain with RV32EMC also, do I need to provide --march=rv32emc along with my compialtion?

jim-wilson, 2020-11-25:
If you used --with-arch=rv32imc when configuring, then -march=rv32imc is the default.  You don't need to pass it.  If you do pass it, that it harmless.  If you have build scripts and multiple compilers, then it might be useful to continue using a -march option in the build scripts, just in case you use a different compiler.  You will then get a link error if the compiler doesn't support -march=rv32imc.  Otherwise, code might be compiled differntly than you expected and you may not notice.

Same is true with rv32emc.

rahulraveendran15-coder, 2020-11-26:
@jim-wilson 
Thank you very much Jim.
One more doubt I have is that what is the difference between "riscv32-unknown-elf-ld"  and "riscv32-unknown-elf-gcc" commands for running

jim-wilson, 2020-11-26:
Depends on what you are running on.  If you are running on a simulator, you can just use "riscv32-unknown-elf-gcc".  If you are running on hardware, you probably need "riscv32-unknown-elf-gcc -nostartfiles -nostdlib" but then you will have to provide your own BSP (board support package).  Most vendors should provide one.  SiFive has freedom-e-sdk for instance.

rahulraveendran15-coder, 2020-11-26:
@jim-wilson 
So what is the use of ""riscv32-unknown-elf-ld",  in both cases you mentioned above the command is "riscv32-unknown-elf-gcc"
when we will use "riscv32-unknown-elf-ld""

jim-wilson, 2020-11-26:
You can add -v to the gcc option to see what options it is passing to ld.  In the simplest case you can just call ld with the object files you need to link together.  In more complicated cases you may need obscure linker options to get the right result.  Generally, gcc has knowledge of the linker options required for various circumstances, so it is easier to use gcc to link.  But if you know what you are doing you can call ld directly to link.

Gcc by default passes a number of *plugin* options to ld.  These are for LTO, and are not needed if not using LTO.

TommyMurphyTM1234, 2022-09-27:
Questions about how to use the toolchain and not issues with the RISC-V GCC toolchain itself.

closed 2022-09-27 by TommyMurphyTM1234