riscv-collab/riscv-gnu-toolchain #786: Problem with gem5

GeorgeMarios, 2020-11-25:
Good evening everyone,
I have configured and built the cross compiler in a Linux machine. I have compiled a “hello world” program too with success. The problem is, when I us the executable in gem5 I have this error:
“panic: panic condition fd < 0 occurred: Failed to open file /lib/ld-linux-riscv32-ilp32d.so.1.”
Any ideas?

jim-wilson, 2020-11-25:
gem5 isn't part of the toolchain and I don't know anything about it.  However, from the error message I can tell that you used a linux compiler, and compiled against the default shared C library.  This can only work if your simulator supports dynamically linked binaries.  qemu does.  Most other simulators don't.  This might work if you static link, e.g. use --static when compiling and linking.  However, it is more likely to work if you use an embedded elf compiler instead of a linux compiler.

GeorgeMarios, 2020-11-26:
Thank you so much for your help. With --static this problem solved. Unfortunately, I can’t run the executable. New error in gem5 is:
“panic: Illegal instruction 0x0000202d at pc 0x0000000000010380:”
Any piece of advice is welcome :)

jim-wilson, 2020-11-27:
You can use objdump -dr to disassemble your program to see what instruction is at that address.  Assuming the endianness is correct that is a jal which should be supported.  if the bytes are swapped, then it is a fld which might  not be supported if the simulator doesn't support FP instructions.  Anyways, I would suggest using objdump to see what the instruction is.

GeorgeMarios, 2020-11-27:
The instruction, which causes the problem, is a jal. I have no idea what to do next.

jim-wilson, 2020-11-27:
This is a compressed jal.  Does gem5 support compressed instructions?  The target address is 2-byte aligned.  Does gem5 supports 2-byte aligned instructions?  It should if it supports compressed instructions.  Maybe you need to pass an option to gem5 to tell if what arch variant you want to simulate?  You probably should be asking someone that knows something about gem5 instead of the toolchain folks, as this isn't a toolchain problem.

GeorgeMarios, 2020-11-27:
Thank you so much!

closed 2020-11-27 by GeorgeMarios