riscv-collab/riscv-gnu-toolchain #787: Compilation on 64bit RISC-V linux (qemu) and cross-compilation (x86_64 -> 64bit RISC-V) of gold linker

advancedwebdeveloper, 2020-11-28:
@jim-wilson , opening this issue since gold is not provided as either a separate Linux package (at least - on Fedora 33 riscv64-unknown-linux-gnu ) or inside "binutils" package.
Debian and OpenSUSE also have their RISC-V ports.

I was trying to compile [under Qemu](https://fedoraproject.org/wiki/Architectures/RISC-V/Installing#Boot_under_QEMU). Building from within "gold" sub-dir. lad to an error, reported by configure, about non-supported arch. (improper triple/naming schema for toolchain's executables?). If I try to use (inside "binutils" directory)
> --enable-gold
I am able to generate a Makefile file - but no "gold" sub-dir. and
> make all-gold
is not building an executable (but does build all the dependencies, as far as I am able to compare with x86_64 Linux target builds, on Linux hosts).

I am unsure if someone was able to compile that before.

Another scenario is to cross-compile - but I am having some general level misunderstanding of --target, --host and --build flags.

advancedwebdeveloper, 2020-11-28:
CC @ianlancetaylor

ianlancetaylor, 2020-11-28:
The gold linker does not yet support RISC/V.

jim-wilson, 2020-11-28:
I am not aware of anyone working on adding the missing RISC-V gold support.  We have only bfd ld and lld RISC-V support at present.

advancedwebdeveloper, 2020-11-29:
CC @sorear 

@Wuyang-Chung , could you tell us if your work
https://riscv.org//wp-content/uploads/2019/03/12.00-Wuyang-Chung-RISC-V-Segmentation-Extension-Proposal.pdf
https://www.youtube.com/watch?v=nkk5HA4LtoY
could be vibrant with the goals of this issue ( re-implementation of https://llvm.org/docs/SegmentedStacks.html on RISC-V)?

@ianlancetaylor , how about if gold could be compiled on RISC-V (configuration scripts would be patched)?
While it wouldn't have practical functionality, in the beginning - it could be of help to focus/contribute into stack segmentation support (and all required stages, which should take place before).
At least this would allow to form a roadmap, for committers.

ianlancetaylor, 2020-11-29:
@advancedwebdeveloper I think it would be fine if someone wants to work on adding RISC/V support to gold.  The configuration is the simplest part of the job, literally just a few lines, so I don't see any reason to do the configuration parts until someone is prepared to work on the rest.

sorear, 2020-11-30:
I have not touched gold.

Wuyang-Chung, 2020-11-30:
@advancedwebdeveloper , my proposal is primary for operating system. If hardware support segmentation, operating system can grow the stack as needed automatically without compiler/application intervention.

lazyparser, 2021-03-03:
> The gold linker does not yet support RISC/V.

Hi @ianlancetaylor 

Thank you for your information. Is there anyone working on supporting RISC-V yet?

If the project is still open for grab, my colleagues from PLCT Lab would like to have a try.

advancedwebdeveloper, 2021-03-03:
@lazyparser , it seems like @ccoutant and @ianlancetaylor could explain more - Cary emailed about his plans, regarding patching gold, for supporting RISC-V arch.
Also there is demand in supporting segmented stacks.

ianlancetaylor, 2021-03-03:
I don't know of anyone adding RISC-V support to gold.

The place to discuss this would the mailing list binutils@sourceware.org.

advancedwebdeveloper, 2021-03-03:
> I don't know of anyone adding RISC-V support to gold.
> 
> The place to discuss this would the mailing list [binutils@sourceware.org](mailto:binutils@sourceware.org).

@lazyparser , please CC to me ( local.tourist.kiev@gmail.com) when you would email into that mailing list

lazyparser, 2021-03-04:
> @lazyparser , it seems like @ccoutant and @ianlancetaylor could explain more - Cary emailed about his plans, regarding patching gold, for supporting RISC-V arch.
> Also there is demand in supporting segmented stacks.

Hi @advancedwebdeveloper Thanks!

lazyparser, 2021-03-04:
> I don't know of anyone adding RISC-V support to gold.
> 
> The place to discuss this would the mailing list [binutils@sourceware.org](mailto:binutils@sourceware.org).

Hi @ianlancetaylor 

Thanks for the information. I'll search the binutils mailinglist and discuss the details there.

lazyparser, 2021-03-04:
> > I don't know of anyone adding RISC-V support to gold.
> > The place to discuss this would the mailing list [binutils@sourceware.org](mailto:binutils@sourceware.org).
> 
> @lazyparser , please CC to me ( [local.tourist.kiev@gmail.com](mailto:local.tourist.kiev@gmail.com)) when you would email into that mailing list

Hi @advancedwebdeveloper 

Sure :)

MaskRay, 2021-03-04:
@ccoutant

> Hi @advancedwebdeveloper
>
> Sure :)

You shall also CC @ccoutant in your binutils@sourceware.org message :)

advancedwebdeveloper, 2021-03-05:
@MaskRay , are you heading to contribute too?

advancedwebdeveloper, 2021-07-15:
Fangrui, ping on you.

MaskRay, 2021-07-15:
No.

SaiVK, 2022-05-12:
Hello everyone
Is it possible to use the _--enable-gold_, _--enable-plugin_ configure while building **_riscv-binutils_**? What I am trying to achieve is the following command to work:
`riscv64-unknown-elf-ld main.o stub.o -plugin=llvm/bins/lib/LLVMgold.so --plugin-opt=save-temps --plugin-opt=O0 --start-group -lc -lm -lgcc --end-group --gc-sections`
I aim to build the Program Dependence Groh of an application, coded across multiple .c files at link time.

Kindly guide me on this

-Sai

jim-wilson, 2022-05-12:
Gold has not been ported to RISC-V yet.  So --enable-gold won't do anything useful.  I don't know anything about LLVMgold.so.

SaiVK, 2022-05-13:
Thanks for clarifying this @jim-wilson 

-Sai

kito-cheng, 2022-05-13:
LLVMgold.so works with ld.bfd for a while, so you actually don't need gold linker if you just want LTO

TommyMurphyTM1234, 2022-09-27:
Main issue is lack of RISC-V support in gold linker.
Not an issue with the RISC-V GCC toolchain itself.

closed 2022-09-27 by TommyMurphyTM1234