riscv-collab/riscv-gnu-toolchain #795: How to add a custom non-standard instruction type format

MauSilvestrini, 2020-12-04:
Good morning to everyone, I want to add a custom non-standard instruction that should have the following type format:

 
[image]

This field format is not new for the RISC-V ISA, it resembles the R4-type format that is used for the encoding of some instructions in the standard extension RV32F for single precision floating point calculation. In that specific case the rs3 source register field has been replaced with the destination register called RD1, in which I want to write something.

I already find out how to encode this new instruction extension, but I’d like to know, as far as possible, the procedure on how to add this new instruction type format to the current RISCV toolchain ISA, in particular which files I have to modify in order to write a piece of code that allows to write onto the RD1 field.

Many thanks for your help.

jim-wilson, 2020-12-04:
Binutils doesn't use instruction formats, except for the .insn pseudo op.  If you want .insn support for your new instruction you might need to change that if the existing r4 instruction format support doesn't work.

Otherwise, instruction formats are irrrelevant, and all that matters are the MASK/MATCH bit masks, and the 'd' 's' 't' etc letters to specify operands.  But if your format is close enough to r4 then you probably don't need a new operand letter.

Nelson1225, 2020-12-07:
FYI, I suppose all instructions can be derived from the existing instruction formats, although some of them might look weird.

Nelson1225, 2020-12-07:
This should help, https://github.com/riscv/riscv-binutils-gdb/pull/184

MauSilvestrini, 2020-12-14:
> Binutils doesn't use instruction formats, except for the .insn pseudo op. If you want .insn support for your new instruction you might need to change that if the existing r4 instruction format [...]

Many thanks for your help Mr Wilson, with this I fixed the problem releted to the riscv-toolchain. Best Regards.

closed 2020-12-14 by MauSilvestrini

TommyMurphyTM1234, 2022-11-10:
This is a list of all issues logged that I can find that relate to adding new instructions to the toolchain:

* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1095
* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1055
* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/837
* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/782
* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1003
* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/795
* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/660
* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/543
* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/341
* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/115

At this stage in the evolution of RISC-V,the majority of instruction addition will be done under the auspices of the relevant groups working on standardising, prototyping, and implementing/enhancing RISC-V extensions and their corresponding instructions. As such, there should be less of a demand for others to do so.

However, where somebody still wants to implement a non stadard custom extension/instruction here are some useful resources:

* GCC internals documentation: https://gcc.gnu.org/onlinedocs/gccint/
* https://hsandid.github.io/posts/risc-v-custom-instruction/
* https://ossg.bcs.org/blog/2020/10/29/adding-an-instruction-to-the-gnu-assembler/
* https://embdev.net/topic/228773
* https://groups.google.com/u/0/a/groups.riscv.org/g/sw-dev/c/sL_OHXYj3LY/m/NOm5Fwa3BAAJ
* https://forums.sifive.com/t/adding-custom-instruction-to-riscv-toolchain/4044/2