riscv-collab/riscv-gnu-toolchain #796: Problem in printing

GeorgeMarios, 2020-12-04:
I have a problem, probably with printf()/puts(). I have downloaded the elf version of gnu-toolchain and run a simple program in gem5. Can you help me in order to solve this? I have search for a gem5 bug too but I found nothing :(

[image]
[image]

kito-cheng, 2020-12-06:
Not sure what your gem5 config and version is, and how is your build steps for `test`? IIRC gem5 was supported for rv64 only before, I am not sure what the current status of gem5, so the possible issue I can guess is you might feed rv32 program to gem5, but it only support rv64, so gem5 wrongly execute the program.

It could be another issue, but it would be gem5 issue rather than riscv-gnu-toolchain issue, I would suggest you asking at gem5's community.

TommyMurphyTM1234, 2022-09-27:
No update/activity in almost two years and seems more likely to be a gem5 rather than a RISC-V GCC toolchain issue.

closed 2022-09-27 by TommyMurphyTM1234