riscv-collab/riscv-gnu-toolchain #797: automatically inserted vsetvli  may cause error

xingmingjie, 2020-12-10:
For such a test case:

```
#include <stdio.h>
#include <riscv_vector.h>

unsigned char a[16] = {1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16};
unsigned short c[8];

int main() {
  vsetvl_e16m1(8);
  vuint16m1_t vb = vzero_u16m1();
  vuint16m1_t vc = vzero_u16m1();
  vsetvl_e8m1(16);
  vuint8m1_t va = vle8_v_u8m1(a);
  vc = vwredsumu_vs_u8m1_u16m1(vc, va, vb);
  vsetvl_e16m1(8);
  vse16_v_u16m1(c, vc);

  for (int i = 0; i < 8; ++i)
    printf("c[%d] = %d\n", i, c[i]);
}
```

The result is correctly when compiled with `-O2`, but incorrectly for `-O0`. The compiler automatically inserted 

`vsetvli x0,x0,e16,m1`

will change vl from 16 to 8, thus `vwredsumu.vs` only calculates 8 elements of a[16].

kito-cheng, 2020-12-10:
Hi @xingmingjie:

Thanks for reporting this, but I can't reproduce that on my local machine, do you mind check which gcc version you are using (e.g. the sha1), I am using this https://github.com/riscv/riscv-gcc/commit/9f8332dce091c93bda60d92d23fbdc8f4253937a and seems work fine both on O0 and O2

xingmingjie, 2020-12-10:
Oops, I haven't update gcc recently. I'm using rvv-intrinsic branch:

`commit 50580bdecb869c8d36bc512987a26dc2c0ae5bd3 (HEAD)`

jim-wilson, 2020-12-10:
That commit id is from early August.  If your rvv-intrinsic riscv-gnu-toolchain is up to date, then you should have a commit id from early September.  Have you done a git pull in the riscv-gnu-toolchain dir recently?  Or maybe you forgot to update submodules after a riscv-gnu-toolchain git update?  What does "git status" say in the riscv-gnu-toolchain dir?  

Although we have more rvv patches on the riscv-gcc branch than are in riscv-gnu-toolchain.  git submodules are a little inconvenient to work with, as every time we update the riscv-gcc rvv branch we have to also remember to update the riscv-gnu-toolchain rvc branch to bring in the gcc patch, and we haven't been doing that regularly.  The commit id Kito mentioned is on the the riscv-gcc riscv-gcc-10.1-rvv-dev branch.  To get that, you would have to cd into riscv-gcc and then checkout the riscv-gcc-10.1-rvv-dev branch.  If you do that, then remember that a git submodule update at the top of the tree will take you back to the old one, unless perhaps we update the riscv-gnu-toolchain rvv-intrinsic branch in the meantime.  Which Kito might do now that I've reminded him.

xingmingjie, 2020-12-11:
Thanks a lot for the detailed introduction. I switched to riscv-gcc-10.1-rvv-dev branch now, and verified the problem has been fixed.  By comparing the generated assembly, I guess it is fixed due to using whole register load/store for stack access.

closed 2020-12-11 by xingmingjie

kito-cheng, 2020-12-11:
Sounds reasonable, we've rewritten stack access at some moment, anyway, welcome for more bug report :)