riscv-collab/riscv-gnu-toolchain #798: Failed when installing with arch rv32i

jzpa, 2020-12-11:
I'm trying to compile it for the rv32i arch
    $ ./configure --prefix=/opt/riscv-rv32i --with-arch=rv32i --with-abi=ilp32
    $ sudo make linux
It failed and finally reported like the image below
[image]
When I try other arches like rv32gc,rv32g, it succeed.
How could I deal with that?

kito-cheng, 2020-12-11:
It seems complain about `sysroff.info` not found, which should located on `riscv-gnu-toolchain/riscv-binutils/binutils/sysroff.info` , so maybe you can go `riscv-gnu-toolchain/riscv-binutils/` and `git status` or just `git reset --hard` there, and try again?

jim-wilson, 2020-12-11:
Linux requires the A (atomic) extension, so "make linux" will not work when given --with-arch=rv32i.  Try rv32ia instead.  The build should fail in glibc though.  The sysroff.info error is odd.  If the sysroff.info file is in the source tree, then maybe there is some odd VPATH problem with make, but that doesn't explain why it worked for rv32gc.  Or maybe you just attached the wrong screen shot?

jzpa, 2020-12-12:
I tried  with arch=rv32ia, and it worked! Thanks for help me dealing with it !
I've tried rv32i again and find out that the error message is 

    ~checking for sysdeps preconfigure fragments... aarch64 alpha arm csky hppa i386 m68k microblaze mips nios2 powerpc riscv glibc requires the A extension
    ~Makefile:234: recipe for target 'stamps/build-glibc-linux-rv32i-ilp32' failed
    ~make: *** [stamps/build-glibc-linux-rv32i-ilp32] Error 1

Sorry, I think I've attached the wrong screen shot.

closed 2020-12-12 by jzpa