riscv-collab/riscv-gnu-toolchain #799: Submodules of rvv-intrinsic branch.

higuoxing, 2020-12-11:
Hi, I want to assemble the riscv-gnu-toolchain with the rvv-extension feature myself.

However, it looks that the `rvv-intrinsic` branch uses totally different versions of submodules. It's difficult for me to figure out where are they from. e.g., `riscv-gcc` is fetched from the `riscv-gcc-10.1-rvv-dev` branch of the `riscv-gcc repo`, `riscv-gdb` is fetched from the `fsf-gdb-9.1-with-sim` branch of the `riscv-binutils-gdb` repo and the `riscv-binutils` is fetched from `rvv-1.0.x-zfh` branch of the `riscv-binutils-gdb` repo (though they don't share exactly the same commit history). Could you please tell me the branch that submodules are developed?

kito-cheng, 2020-12-11:
>  Could you please tell me the branch that submodules are developed?
Basically your guess is right, seems like I should mention the branch info so that other contributor could easier to contribute.

higuoxing, 2020-12-11:
> > Could you please tell me the branch that submodules are developed?
> 
> Basically your guess is right, seems like I should mention the branch info so that other contributor could easier to contribute.

Great! Thanks a lot!

jim-wilson, 2020-12-11:
This is one thing that git submodules makes more complicated than it shoul be, as submodules are identified by commit id, and there is no easy way to map a commit id back to a branch.

binutils and gdb share the same git repo, but they are maintained by different teams, and make releases at different times of the year, so they have separate release branches.  The FSF doesn't like it when people make releases from development trees, so we are using releases branches, and that means we need to check out separate branches from the same git tree for binutils and gdb sources.

fanghuaqi, 2020-12-14:
Hi @jim-wilson @kito-cheng , I have a suggestion about mention the submodule branches used in different branch of this repo, maybe the wiki page can be used to let people know which branches are actively developed, and used for which feature, and its submodule's branch is based on which branch.

PR #819 (2021-01-11): Specify branch info in .gitmodules.
This PR adds branch info in .gitmodules, which helps resolve #799
commit f6d57520 (2021-01-11): Specify branch info in .gitmodules.

This resolves #799
commit 4e94cf88 (2022-05-09): Specify branch info in .gitmodules.

This resolves #799
closed 2021-01-11 by higuoxing