riscv-collab/riscv-gnu-toolchain #800: can't build on apple silicon mac (arm64)

Warden0317, 2020-12-11:
ld: symbol(s) not found for architecture arm64
[image]

jim-wilson, 2020-12-11:
GCC needs some build/host OS support, and there is no gcc support for darwin (aka MacOS) running on arm.  So this isn't going to work.  Someone needs to port GCC to aarch64-darwin first.  And then we need to pull those patches down into riscv-gnu-toolchain.  Apple no longer does GNU toolchain work, so I doubr that Apple will provide the missing gcc support.  Some volunteer will need to do the work.  So it could be a while before this support is available.

The build/host support is a lot easier than the target support though.  If you are motivated you could try adding this yourself.    In riscv-gcc, see gcc/config.gcc and search for darwin.  Note the x86_64-darwin and powerpc64-darwin stuff.  You need to add something similar for aarch64-darwin.  You can try just copying the x86_64 files from the gcc/config/i386 dir to the gcc/config/aarch64 dir, changing variable names and file names as necessary, and it might just work if darwin didn't change for the aarch64 port.

ChoungJX, 2020-12-13:
You can running in Rosetta2. Just use homebrew to install it.
`arch --x86_64 /usr/local/bin/brew tap riscv/riscv`
`arch --x86_64 /usr/local/bin/brew install riscv-tools`

Then add link: `PATH=$PATH:/usr/local/opt/riscv-gnu-toolchain/bin`

ilg-ul, 2020-12-13:
I did not try building the RISC-V GCC cross-toolchain on Apple Silicon yet, so I'm not sure how well it goes, but anyway you should try it with the available Apple clang, since, as Jim mentioned, GCC is not yet available and it might not be for a while:

- https://gcc.gnu.org/bugzilla/show_bug.cgi?id=96168

Given the uncertain future of GCC on macOS, these days I'm experimenting with Apple clang 12, for now on an Intel Mac, since I don't have an Arm one yet, but I guess they are very similar. 

So far I managed to compile, with minor issues, all packages that I needed, thus I'm confident that migrating the native compiler in my build scripts from GCC to clang is possible.

fabiojmendes, 2020-12-24:
The latest GCC available on [macports](https://ports.macports.org/port/gcc-devel/summary) (devel branch) works fine on arm64, I can confirm that.

```
$ file $(which gcc-mp-devel) 
/opt/local/bin/gcc-mp-devel: Mach-O 64-bit executable arm64

$ gcc-mp-devel --version
gcc-mp-devel (MacPorts gcc-devel 11-20201205_0) 11.0.0 20201205 (experimental)
Copyright (C) 2020 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

$ gcc-mp-devel -dumpmachine
arm64-apple-darwin20
```

jim-wilson, 2020-12-25:
We still need patches for riscv-gnu-toolchain/riscv-gcc to bulid on arm64-apple-darwin20.  There is some host support that is required in the RISC-V gcc sources that get compiled as part of the RISC-V toolchain build.  It may be a while before those patches get upstreamed to the FSF GCC and then pulled down into riscv-gnu-toolchain.

GuangsZuo, 2021-01-26:
@ChoungJX 's method works, for clarifying: you need to use arch --x86_64 install another homebrew. https://stackoverflow.com/a/64997047/3013527 here is a reference.

robertlipe, 2021-02-20:
I found a kludge that worked well enough it was codified for home-brew,

https://github.com/riscv/homebrew-riscv/issues/47

It just disabled mmap on precompiled headers on M1.

SteveCoul, 2021-03-27:
Attached is the patch I use for my toolchain ( I'm switching over to using this project rather than my own but not there yet ).

diff -Naur gcc-10.2.0/gcc/config/host-darwin.c gcc-10.2.0.new/gcc/config/host-darwin.c
--- gcc-10.2.0/gcc/config/host-darwin.c	2020-07-23 01:35:17.000000000 -0500
+++ gcc-10.2.0.new/gcc/config/host-darwin.c	2021-03-07 19:58:50.000000000 -0600
@@ -22,6 +22,8 @@
 #include "coretypes.h"
 #include "diagnostic-core.h"
 #include "config/host-darwin.h"
+#include "hosthooks.h"
+#include "hosthooks-def.h"
 
 /* Yes, this is really supposed to work.  */
 static char pch_address_space[1024*1024*1024] __attribute__((aligned (4096)));
@@ -75,3 +77,5 @@
 
   return ret;
 }
+const struct host_hooks host_hooks = HOST_HOOKS_INITIALIZER;
+

fengxiaohu, 2022-06-14:
So is there any way I can try to build on my m1 macbook.
I suppose 2 ways.
1.Use docker 
2. Use virtual machine like Parallels Desktop

fengxiaohu, 2022-06-15:
> I also have successful builds on Apple SIlicon, but with the GNU 11.3 sources plus some patches:
> 
> * https://github.com/xpack-dev-tools/riscv-none-elf-gcc-xpack/

thx,I will try to use  your method

ilg-ul, 2022-06-15:
The patch for 11.3 is small:

- <https://github.com/xpack-dev-tools/riscv-none-elf-gcc-xpack/blob/xpack/patches/gcc-11.3.0.patch.diff>

12.1 builds without patches.

TommyMurphyTM1234, 2022-11-11:
Hi @cmuellner and @kito-cheng - do you think that this can be closed?

cmuellner, 2022-11-11:
One idea would be to close MacOS issues and redirect people to https://github.com/riscv-software-src/homebrew-riscv

cmuellner, 2022-11-11:
Btw, the current version of MacOS is 13.0.1. And even homebrew-riscv does not build with that (at least I did not manage to get it building within 2 hours because of tons of configuration issues, dependency issues and path issues).

TommyMurphyTM1234, 2022-11-12:
> One idea would be to close MacOS issues and redirect people to https://github.com/riscv-software-src/homebrew-riscv

Thanks @cmuellner - I'll do this for any open MacOS build issues.

closed 2022-11-12 by TommyMurphyTM1234