riscv-collab/riscv-gnu-toolchain #803: compile error: ld: -shared not supported

Dream-math, 2020-12-24:
Hi, when I run:
``` C
/home/hye/soft/riscv/_install/bin/riscv64-unknown-elf-g++ dll.c dll2.c dll3.c -shared -fPIC -o dll.so
```
the error comes:
```
/home/hye/soft/riscv/_install/bin/../lib/gcc/riscv64-unknown-elf/10.1.0/../../../../riscv64-unknown-elf/bin/ld: -shared not supported
collect2: error: ld returned 1 exit status
```
Anyone knows how to solve this problem? Thanks!

jim-wilson, 2020-12-25:
The embedded elf toolchain has no shared library support.  If you are just building firmward like a boot loader it is possible to work around this, but building dlls or dynamically linked applications won't work without shared library support in newlib which is missing.

Shared libraries are supported by the linux toolchain.  If you just want to build applications to run on a simulator, then use a linux toolchain and qemu.  Most other simulators (like spike) can't run dynamically linked programs.

Dream-math, 2020-12-25:
Thanks, in fact I am building applications to run on a simulator, since I do not have a physical riscv processor. I get a module which is generated by llvm for riscv architecture.  The module will be saved as a dll, so I can write a program to call it to verify my module. Could you give me some advice, what compiler and environment do I need to verify my module. I am new in riscv. Thank you very much!

jim-wilson, 2020-12-25:
You want to build a linux toolchain instead of an embedded elf toolchain.  You can use --enable-linux when configuring the toolchain.  Or if you have access to a recent linux distro like Ubuntu 20.04 you can just apt install (or whatever) a riscv64-linux toolchain.

You will need to use qemu to run code using shared libraries.  riscv-gnu-toolchain will build qemu with "make build-qemu".  Or again, if on a recent enough linux distro you can just apt install (or whatever) a user-mode qemu wtih riscv64-linux support.  The RISC-V support may be in the qemu "misc" package.

Dream-math, 2020-12-25:
Hi, @jim-wilson thanks for your advice, the error is now:
```
riscv64-unknown-linux-gnu/bin/ld: /tmp/tmpmjrs6kwu/lib0.o: can't link soft-float modules with double-float modules
```
my configure is:
```
riscv64-unknown-linux-gnu-gcc --verbose
Using built-in specs.
COLLECT_GCC=riscv64-unknown-linux-gnu-gcc
COLLECT_LTO_WRAPPER=/home/hye/soft/riscv/new_install/libexec/gcc/riscv64-unknown-linux-gnu/10.1.0/lto-wrapper
Target: riscv64-unknown-linux-gnu
Configured with: /home/hye/soft/riscv/riscv-gnu-toolchain/riscv-gcc/configure --target=riscv64-unknown-linux-gnu --prefix=/home/hye/soft/riscv/new_install --with-sysroot=/home/hye/soft/riscv/new_install/sysroot --with-system-zlib --enable-shared --enable-tls --enable-languages=c,c++,fortran --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libsanitizer --disable-nls --disable-bootstrap --src=.././riscv-gcc --enable-multilib --with-abi=lp64d --with-arch=rv64imafdc --with-tune=rocket 'CFLAGS_FOR_TARGET=-O2   -mcmodel=medlow' 'CXXFLAGS_FOR_TARGET=-O2   -mcmodel=medlow'
Thread model: posix
Supported LTO compression algorithms: zlib zstd
gcc version 10.1.0 (GCC)
```
thanks for your help.

kito-cheng, 2020-12-25:
It seems like multilib issue, what arch and abi you using?

Dream-math, 2020-12-25:
Thanks, @kito-cheng, sorry I don't get it, the lib0.o is a file generated by Apache TVM, the tvm target string is 
```
llvm -mtriple=riscv64-unknown-linux-gnu -mcpu=rocket-rv64
```
and my host machine is 
```
x86_64-unknown-linux-gnu
```

kito-cheng, 2020-12-25:
Using `riscv64-unknown-elf-readelf -hA` could check the what ABI and what arch and then add corresponding -march and -mabi to your gcc command.

Dream-math, 2020-12-25:
@jim-wilson @kito-cheng thank all of you, it finally works!

Dream-math, 2021-01-04:
Hi, @kito-cheng I have a question about qemu-riscv64, when I run
```
qemu-riscv64 -cpu rv64,x-v=true,vlen=256,elen=64,vext_spec=v1.0 vec_mul
```
It comes with
```
qemu-riscv64: Unsupported vector spec version 'v1.0'
```
In this page https://lists.riscv.org/g/tech-vector-ext/message/364, I see that you execute correctly.
Could you tell me which version qemu do you use, I can not find sifive qemu anywhere, thanks!

kito-cheng, 2021-01-04:
@Dream-math you can using `v5.1.0-rvv-zfh-pmp` branch from SiFive's qemu repo.

https://github.com/sifive/qemu/tree/v5.1.0-rvv-zfh-pmp

Dream-math, 2021-01-05:
@kito-cheng thanks for your reply, but when I finished building qemu, the error becomes
```
qemu-riscv64: can't apply global rv64-riscv-cpu.x-v=true: Property '.x-v' not found
```
In the process of building sifive qemu, it has a problem
```
qemu/include/disas/capstone.h:6:10: fatal error: capstone.h: No such file or directory
 #include <capstone.h>
```
I fix it by replacing
```
#include <capstone.h>
```
with
```
#include <capstone/capstone.h>
```
Could you tell me if I missed something, how to solve it, thanks!

kito-cheng, 2021-01-05:
> qemu-riscv64: can't apply global rv64-riscv-cpu.x-v=true: Property '.x-v' not found
it should be comma (`,`) rather than dot (`.`)

> qemu/include/disas/capstone.h:6:10: fatal error: capstone.h: No such file or directory
> #include <capstone.h>
I guess maybe you are using same directory and then switch to another repo/branch? if so I would suggest you run `git submodule sync --recursive&&  git submodule update --recursive` and try again.

Dream-math, 2021-01-05:
Thanks, @kito-cheng, in fact my command is:
```
qemu-riscv64 -cpu rv64,x-v=true,vlen=256,elen=64,vext_spec=v1.0 vec_mul
```
it is indeed comma, but the error message displays dot. And when I try to compile riscv-gnu-toolchain rvv-intrinsic version,
there is a error:
```
riscv-gnu-toolchain/riscv-gcc/isl/missing: line 81: aclocal-1.15: command not found
```
I have installed aclocal, I can not understand why it requires aclocal-1.15, my aclocal is
```
$aclocal --version
aclocal (GNU automake) 1.16.1
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv2+: GNU GPL version 2 or later <https://gnu.org/licenses/gpl-2.0.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.

Written by Tom Tromey <tromey@redhat.com>
       and Alexandre Duret-Lutz <adl@gnu.org>.
```
Thanks!

closed 2021-03-03 by lazyparser

Aashir1234, 2025-01-17:
this is my toolchain :
/Andestech/toolchains/nds32le-elf-mculib-v5//nds32le-elf-mculib-v5/bin/riscv32-unknown-elf-gcc
i want to create sharedobject of whole production code base 
but there is no support for --shared in my tool chain 
when i try to create a shared object i get this error :
/home/aashir/Andestech/toolchains/nds32le-elf-mculib-v5/nds32le-elf-mculib-v5/bin/../lib/gcc/riscv32-elf/10.4.0/../../../../riscv32-elf/bin/ld: -shared not supported
Is there is any possible way to add support for --shared ?
I am a beginner in this field

TommyMurphyTM1234, 2025-01-17:
> this is my toolchain : /Andestech/toolchains/nds32le-elf-mculib-v5//nds32le-elf-mculib-v5/bin/riscv32-unknown-elf-gcc i want to create sharedobject of whole production code base but there is no [...]

Please read the full issue discussion and search for related open and closed issue discussions for an explanation for why the bare-metal toolchain (which cannot presuppose the existence of an appropriate OS environment and link loader etc.) can't support the creation and use of shared libraries. Your question isn't one specific to RISC-V, never mind the `riscv-gnu-toolchain`, so would be better posed in another more suitable forum. E.g. upstream GCC mailing lists, StackOverflow, Reddit etc.

lygstate, 2025-12-30:
@jim-wilson 
```
RISC-V: Disable shared library support for embedded elf.

	ld/
	* emulparams/elf32lriscv-defs.sh (GENERATE_SHLIB_SCRIPT): Move inside
	case on $target, and don't set for riscv*-elf targets.
	(GENERATE_PIE_SCRIPT): Likewise.

```

getting riscv64-unknown-elf can not building u-boot and opensbi, can we renable it now?
anything prevent it to re-enable it?

The original reason is:
```

Runnning the linker testsuite with a proper cross compiler in place shows over
160 unexpected failures for riscv32-elf and riscv64-elf.  These are all tests
trying to build and use shared libraries, but getting relocation errors on
newlib startfiles, because we don't have shared library support there.  I don't
think we need it, at least not at the moment.  So this just stops setting
GENERATE_SHLIB_SCRIPT and GENERATE_PIE_SCRIPT for embedded elf targets.

This was tested with riscv{32,64}-{elf,linux} targets on check-binutils,
check-gas, and check-ld.  There were no changes to the linux results.  The
elf ld results dropped by over 150 and the other testsuites had no change.

I also tested this by verifying that ld -shared gave the expected result with
all four linkers.

Committed.

```

According to https://sourceware.org/pipermail/binutils/2018-January/101045.html

TommyMurphyTM1234, 2025-12-30:
> getting riscv64-unknown-elf can not building u-boot and opensbi, can we renable it now?
> anything prevent it to re-enable it?

I don't understand this. Please clarify. E.g. provide clear and complete details on how to recreate whatever issue you are experiencing.

Have you read the full issue discussion for an explanation about why there may be issues with supporting shared libraries with the bare metal toolchain?

lygstate, 2025-12-30:
> I don't understand this. Please clarify. E.g. provide clear and complete details on how to recreate whatever issue you are experiencing.
>
> Have you read the full issue discussion for an explanation [...]

For opensbi:
```bash
 make -C opensbi clean
Makefile:207: *** Your linker does not support creating PIEs, opensbi requires this..  Stop.
```
For uboot:
```bash
$ make -C u-boot -j65
make: Entering directory '/c/work/xxx/xxx/u-boot'
  UPD     include/generated/timestamp_autogenerated.h
  CC      common/version.o
  AR      common/built-in.o
  LD      u-boot
riscv64be-unknown-elf-ld.bfd: -pie not supported
make: *** [Makefile:1826: u-boot] Error 1
make: Leaving directory '/c/work/xxx/xxx/u-boot'
```

But both opensbi and uboot are bare-metal firmware.