riscv-collab/riscv-gnu-toolchain #804: "mcountinhibit" CSR - Name alias (in assembly code)

stnolting, 2020-12-25:
While implementing the `mcountinhibit` CSR to a [custom core](https://github.com/stnolting/neorv32) I noticed that this alias is not understood by the assembler:

When compiling `csrw mcountinhibit, x0` I get:

> Error: unknown CSR `mcountinhibit'

When I use the actual address of that CSR as immediate (`csrw 0x320, x0`) everything works fine, but when checking the generated assembly code I find:

> `csrw mucounteren, zero`

Is `mucounteren` an "old" name for `mcountinhibit` or where does that come from?

I am using RISC- GCC 10.1.0 and binutils 2.34 (if that is relevant). Maybe my toolchain is out of date?!

aswaterman, 2020-12-28:
It looks like your assembler or disassembler is targeting an old, incompatible version of the privileged architecture spec (1.9.1 vs. 1.10 or newer?).  mcounteren has been at address 306 for about 4 years now.

kito-cheng, 2020-12-28:
@Nelson1225 has added that, and it included in binutils 2.35, so I would suggest you upgrade your toolchain :)

stnolting, 2020-12-29:
Oh ok 😅 
Thanks for the hint!

closed 2020-12-29 by stnolting