riscv-collab/riscv-gnu-toolchain #805: riscv32-unknown-elf-g++ does not use my start.S with -lstdc++ option.

vagrantxiao, 2020-12-28:
I am using riscv32-unknown-elf-g++ to compile my code. When I add -lstdc++ option, I find my program does not run. I use riscv32-unknown-elf-objdump and find it generates its own _start function and does not initialize the sp. Is there a way to force the tool chain to use my start.S instead of generates its own _start entry function.

The disassemble code with -lstdc++

Disassembly of section .memory:

```
00000000 <_start>:
       0:	00002197          	auipc	gp,0x2
       4:	fdc18193          	addi	gp,gp,-36 # 1fdc <__global_pointer$>
       8:	00002517          	auipc	a0,0x2
       c:	7d450513          	addi	a0,a0,2004 # 27dc <_end>
      10:	00002617          	auipc	a2,0x2
      14:	7cc60613          	addi	a2,a2,1996 # 27dc <_end>
      18:	40a60633          	sub	a2,a2,a0
      1c:	00000593          	li	a1,0
      20:	3fd000ef          	jal	ra,c1c <memset>
      24:	00001517          	auipc	a0,0x1
      28:	af450513          	addi	a0,a0,-1292 # b18 <__libc_fini_array>
      2c:	2a5000ef          	jal	ra,ad0 <atexit>
      30:	345000ef          	jal	ra,b74 <__libc_init_array>
      34:	00012503          	lw	a0,0(sp)
      38:	00410593          	addi	a1,sp,4
      3c:	00000613          	li	a2,0
      40:	100010ef          	jal	ra,1140 <main>
      44:	2a10006f          	j	ae4 <exit>

```

The dissemble code with -nostdlib
```
Disassembly of section .memory:

00000000 <_start>:
       0:	00000093          	li	ra,0
       4:	00000113          	li	sp,0
       8:	00000193          	li	gp,0
       c:	00000213          	li	tp,0
      10:	00000293          	li	t0,0
      14:	00000313          	li	t1,0
      18:	00000393          	li	t2,0
      1c:	00000413          	li	s0,0
      20:	00000493          	li	s1,0
      24:	00000513          	li	a0,0
      28:	00000593          	li	a1,0
      2c:	00000613          	li	a2,0
      30:	00000693          	li	a3,0
      34:	00000713          	li	a4,0
      38:	00000793          	li	a5,0
      3c:	00000813          	li	a6,0
      40:	00000893          	li	a7,0
      44:	00000913          	li	s2,0
      48:	00000993          	li	s3,0
      4c:	00000a13          	li	s4,0
      50:	00000a93          	li	s5,0
      54:	00000b13          	li	s6,0
      58:	00000b93          	li	s7,0
      5c:	00000c13          	li	s8,0
      60:	00000c93          	li	s9,0
      64:	00000d13          	li	s10,0
      68:	00000d93          	li	s11,0
      6c:	00000e13          	li	t3,0
      70:	00000e93          	li	t4,0
      74:	00000f13          	li	t5,0
      78:	00000f93          	li	t6,0
      7c:	00020137          	lui	sp,0x20
      80:	deadc1b7          	lui	gp,0xdeadc
      84:	eef18193          	addi	gp,gp,-273 # deadbeef <_end+0xdeada0c3>
      88:	00018213          	mv	tp,gp
      8c:	33d000ef          	jal	ra,bc8 <main>
      90:	10000537          	lui	a0,0x10000
      94:	04400593          	li	a1,68
      98:	04f00613          	li	a2,79
      9c:	04e00693          	li	a3,78
      a0:	04500713          	li	a4,69
      a4:	00a00793          	li	a5,10
      a8:	00b52023          	sw	a1,0(a0) # 10000000 <_end+0xfffe1d4>
      ac:	00c52023          	sw	a2,0(a0)
      b0:	00d52023          	sw	a3,0(a0)
      b4:	00e52023          	sw	a4,0(a0)
      b8:	00f52023          	sw	a5,0(a0)
      bc:	20000537          	lui	a0,0x20000
      c0:	075bd5b7          	lui	a1,0x75bd
      c4:	d1558593          	addi	a1,a1,-747 # 75bcd15 <_end+0x75baee9>
      c8:	00b52023          	sw	a1,0(a0) # 20000000 <_end+0x1fffe1d4>
      cc:	00100073          	ebreak

```

jim-wilson, 2020-12-30:
_start comes from crt0.o not libstdc++.  You can disable use of the default crt0.o by using --nostartfiles.  However, if you do that, you will need to replace all of the default start files, which inclucdes crt0.o, but also includes crtbegin.o and crtend.o which are needed to make C++ static constructors and other C++ features work.

Another alternative if you to add a -v option to your gcc command to get the link command, and then write your own link command, replacing crt0.o with your own _start definition.

The default newlib support really only works on simulators.  If you want to run on hardware, you should use a BSP for that hardware, and the bsp will contain its own crt0.o that initializes sp appropriately for that hardware.

lazyparser, 2021-03-03:
Hi @vagrantxiao 

is this issue resovled so we can close it now?

vagrantxiao, 2021-03-03:
Hi Wei Wu,

The problem has been resolved. Thanks for your follow-up!

Best,
Yuanlong

closed 2021-03-04 by lazyparser