riscv-collab/riscv-gnu-toolchain #807: ERROR When Compiling C File

Louisyoung056, 2020-12-30:
Hi,

I followed instructions on https://github.com/riscv/riscv-gnu-toolchain, installed the newest riscv-gnu-toolchain on my CentOS7 virtual machine, and successfully built the Newlib cross-compiler. According to the README, I should then be able to use riscv64-unknown-elf-gcc and its cousins. But it shows error message like this when I try to compile my C program(test.c) into machine code:

Commands:
`riscv64-unknown-elf-gcc -mno-relax -march=rv32imc -mabi=ilp32 -nostartfiles -std=gnu11 -mstrict-align -mno-div -O0 crt.s test.c -T linker.s -o program.elf`

Error message:
```
/opt/riscv64/lib/gcc/riscv64-unknown-elf/10.2.0/../../../../riscv64-unknown-elf/bin/ld: /opt/riscv64/lib/gcc/riscv64-unknown-elf/10.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-rand.o): **ABI is incompatible with that of the selected emulation:
  target emulation `elf64-littleriscv' does not match `elf32-littleriscv'**
/opt/riscv64/lib/gcc/riscv64-unknown-elf/10.2.0/../../../../riscv64-unknown-elf/bin/ld: failed to merge target specific data of file /opt/riscv64/lib/gcc/riscv64-unknown-elf/10.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-rand.o)
/opt/riscv64/lib/gcc/riscv64-unknown-elf/10.2.0/../../../../riscv64-unknown-elf/bin/ld: /opt/riscv64/lib/gcc/riscv64-unknown-elf/10.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-impure.o): ABI is incompatible with that of the selected emulation:
  target emulation `elf64-littleriscv' does not match `elf32-littleriscv'
/opt/riscv64/lib/gcc/riscv64-unknown-elf/10.2.0/../../../../riscv64-unknown-elf/bin/ld: failed to merge target specific data of file /opt/riscv64/lib/gcc/riscv64-unknown-elf/10.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-impure.o)
collect2: error: ld returned 1 exit status
```
Is there something wrong with the arguments I added? Any help would be appreciated. Thanks!

LevyHsu, 2020-12-30:
You compiled a 64bit "riscv64-unknown-elf-gcc"

Louisyoung056, 2020-12-30:
> You compiled a 64bit "riscv64-unknown-elf-gcc"

Could you please elaborate?
Does this mean that I shouldn't use "-march=rv32imc -mabi=ilp32" ?

kito-cheng, 2020-12-30:
The issue might cause by your gcc only support riscv64, and might cause by multi-lib supporting not enabled.

But I am not sure because that's need your gcc configure to kwon what happen, do you mind post your gcc configure, it could dump via `riscv64-unknown-elf-gcc -v` command.

If it's cause by you didn't enable multi-lib, solution is build your toolchain with `--enable-multilib`, or build with `--with-arch=rv32imc --with-abi=ilp32`.

LevyHsu, 2020-12-30:
Try re-compile a 32bit one:
First cd into your toolchain build folder,
`make clean`
`./configure --prefix=/opt/riscv **--with-arch=rv32imc --with-abi=ilp32**`
`make -j<>`
Then call **riscv32-unknown-elf-gcc**, See if it works.

Louisyoung056, 2020-12-30:
> The issue might cause by your gcc only support riscv64, and multi-lib supporting not enabled.
> 
> But I am not sure because that's need your gcc configure to kwon what happen, do you mind post your gcc configure, it could dump via `riscv64-unknown-elf-gcc -v` command.
> 
> If it's cause by you didn't enable multi-lib, solution is build your toolchain with `--enable-multilib`, or build with `--with-arch=rv32imc --with-abi=ilp32`.

Thank you for your reply.
Here is my gcc configuration:
`$ riscv64-unknown-elf-gcc -v
Using built-in specs.
COLLECT_GCC=riscv64-unknown-elf-gcc
COLLECT_LTO_WRAPPER=/opt/riscv/libexec/gcc/riscv64-unknown-elf/10.2.0/lto-wrapper
Target: riscv64-unknown-elf
Configured with: /home/designer/riscv-gnu-toolchain/riscv-gcc/configure --target=riscv64-unknown-elf --prefix=/opt/riscv --disable-shared --disable-threads --enable-languages=c,c++ --with-system-zlib --enable-tls --with-newlib --with-sysroot=/opt/riscv/riscv64-unknown-elf --with-native-system-header-dir=/include --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libgomp --disable-nls --disable-tm-clone-registry --src=.././riscv-gcc --disable-multilib --with-abi=lp64d --with-arch=rv64imafdc --with-tune=rocket 'CFLAGS_FOR_TARGET=-Os   -mcmodel=medlow' 'CXXFLAGS_FOR_TARGET=-Os   -mcmodel=medlow'
Thread model: single
Supported LTO compression algorithms: zlib
gcc version 10.2.0 (GCC)  `

It seems that I disabled multilib. Should I rebuild again like this:
`./configure --prefix=/opt/riscv --enable-multilib `

Louisyoung056, 2020-12-30:
> Try re-compile a 32bit gcc:
> First `cd` into your toolchain build folder,
> `make clean`
> `./configure --prefix=/opt/riscv --with-arch=rv32imc --with-abi=ilp32`
> `make -j<core_numbers>`
> Then call **riscv32-unknown-elf-gcc**, See if it works.

Thanks for the reply!
I will give it a try!

LevyHsu, 2020-12-30:
Either build with multilib for different arch and abi support, in case you need something different, 
or just build the 32bit one you currently need. 

Both methods should works.

Louisyoung056, 2020-12-31:
> Either build with multilib for different arch and abi support, in case you need something different,
> or just build the 32bit one you currently need.
> 
> Both methods should works.

Thanks for the advice. Problem solved using `--enable-multilib` !

closed 2020-12-31 by Louisyoung056