riscv-collab/riscv-gnu-toolchain #808: tools binary package

agrobman, 2020-12-30:
Is  precompiled binary package for linux available?

I've found one on ubuntu, but seems it's outdated and not multilib. I had to spend many hours to compile this one plus 15GB disc space. (I'm not a programmer)  I believe community can benefit greatly if you could create such package. At least for embedded elf output version with multilib support, which is more universal.

jim-wilson, 2020-12-30:
We don't have any staff, only overworked volunteers.  I would suggest the ubuntu pre-built one.  Ubuntu and other desktop distros only support riscv64-linux though.  If you need riscv32-linux, then yes you will need to build it.  Note that the riscv32-linux glibc support was only upstreamed recently, after the last glibc release, and may not be compatible with the riscv32-linux toolchain you will get from riscv-gnu-toolchain which is still using the old riscv32 glibc port.  Unfortunately, there is no one supporting riscv32-linux upstream so it may not be stable.  OpenEmbedded/Yocto does have riscv32-linux support though, and is probably the easiest way to build one, though this is probably still using the old glibc port.  Basically, don't use riscv32-linux unless you are serious, and willing to fix problems that you might run into.

kito-cheng, 2020-12-31:
There is some pre-built package is built by github action [1], I guess it would be useful, but we don't have enough engineer resource to do that, so in case you are interesting to contribute on that you can take a look on there [2].

[1] https://github.com/riscv/riscv-gnu-toolchain/actions?query=workflow%3APackaging
[2] https://github.com/riscv/riscv-gnu-toolchain/tree/master/.github

agrobman, 2021-01-01:
I was thinking that linux tools version is very specific for few folks as there is no many HW RiSCV platforms are built, but much  more embedded are in use/development. Also multilib one can be pretty universal - supporting all or almost all flavors RISCV designs - 64/32 bits plus any extensions. 

Ubuntu packet versions are very specific and does not work with 32 bits designs - pretty much useless for majority embedded users.

I guess it should not take a lot of time to build that multilib 'newlib' package for somebody who is in the theme ... .
Since you guys are the source of these tools de-facto, it more reasonable if you create this one ... to help many like me.
In fact, I'd built the package first time in my life and had to run make multiple times since my ubuntu was missing many packages and it took some web research to figure this out for each trial.

I would create this package by myself, but I'm not a programmer and know nothing about building tools/packages for Linux.  

BTW, Sorry for my naive question, but is it possible to build universal toolset to support both Linux and embedded Elf to reduce users confusion? 

Another strange thing was that I had coremark benchmark compilation problem with ubuntu package - something with
'uint32' vs 'u_int32' types. 'uint32' works with multilib compiler and doesn't with ubuntu ?

jim-wilson, 2021-01-01:
Unfortunately it is far more complicated than you realize.  All current extensions and subextensions leads to millions of multilibs which is impractical to build.  There is the issue of what we build for.  There is more than one linux distro, and more than one version of them, so a single toolchain isn't going to work.  But the biggest problem is that there is more than one way to build a linux system, and a cross compiler is not going to work with your target linux system unless it is using the same tools versions configured in a compatible way.  riscv-gnu-toolchain doesn't support this.  It can't.  riscv-gnu-toolchain is useful for building programs to run on a simulator to play with.  But it is useless for real linux hardware.  If you are serious about riscv linux you need to stop looking at riscv-gnu-toolchain as it is the wrong way to build a compiler for what you are doing.  You instead need to use the compiler that your target linux system wants.  Try lookiing at openembedded/yocto instead.  It will build a cross compiler for you that will work with the target linux system.

There are a lot of differences between embedded elf and linux toolchains.  Different C libraries.  Different default compiler, assembler, and linker options.  Possibly different built in types like wchar_t.  I'm not aware of anyone that supports a GNU toolchain that supports both at the same time.

Different programs and different C libraries may define different types.  That is outside the scope of the toolchain.  The only thing I can comment on is that the ISO C language defines a uint32_t type defined in the stdint.h header file.  If you want portable code, stick to header files and types defined by the ISO C standard.

agrobman, 2021-01-02:
Jim, thanks for clarification, actually I was looking for pre-built embedded - elf target version, not linux one. I believe that majority of RISCV applications are for embedded systems, not for Linux, at least at current moment. Once somebody creates linux capable RISCV processor and system for it, then these people should be capable to build Linux targeted toolchain...
I guess such projects can afford efforts to build these tools from sources. I was talking about "small" folks support.. 

BTW, the ubuntu compiler version complained about uint32_t type, suggesting something like 'u_int32_t' ..

ilg-ul, 2021-01-02:
> I was looking for pre-built embedded - elf target version

https://github.com/xpack-dev-tools/riscv-none-embed-gcc-xpack/releases

The 10.x version will be released soon.

agrobman, 2021-01-02:
thanks Liviu,

what GCC version your gcc is based on? 

How to make community to know of your RISCV XPACK?

ilg-ul, 2021-01-02:
> what GCC version your gcc is based on?

The v8.3.0-2.3 binaries were built with exactly the same sources as the SiFive GCC 8.3 release. The new release will use the ones in the SiFive GCC 10.1 binaries.

For full details please follow the **Continue reading** links.

> How to make community to know of your RISCV XPACK?

Google for 'riscv gcc'? Read the SiFive download page?

jim-wilson, 2021-01-02:
SiFive has embedded elf toolchain binaries on its web site, and pointers to 3rd party toolchains, including Liviu's toolchain.

As for the uint32_t problem, it is hard to give a good answer without a proper bug report, including enough info to reproduce.  But I would guess that the problem is that recent versions of gcc have some spell checking support for errors, you forgot to include stdint.h, and you happen to have a similarly named type defined in a random header file.  If you really wanted uint32_t, then ignore the error and add the missing stdint.h include.
```
rohan:2049$ cat tmp.c
typedef unsigned int u_int32_t;
uint32_t a;
rohan:2050$ riscv32-unknown-elf-gcc -c tmp.c
tmp.c:2:1: error: unknown type name 'uint32_t'; did you mean 'u_int32_t'?
    2 | uint32_t a;
      | ^~~~~~~~
      | u_int32_t
rohan:2051$ cat tmp2.c
#include <stdint.h>
typedef unsigned int u_int32_t;
uint32_t a;
rohan:2052$ riscv32-unknown-elf-gcc -c tmp2.c
rohan:2053$ 
```

kevinpark1217, 2021-01-12:
https://github.com/riscv/riscv-gnu-toolchain/issues/747#issuecomment-758513770
Above link is also a related issue.

I was wondering if someone could point me in the correct direction for package naming convention and maybe couple of widely used configurations.

TommyMurphyTM1234, 2022-09-27:
No update/activity in over a year and the question about options for obtaining prebuilt binary versions of the RISC-V GCC toolchain has been answered.

closed 2022-09-27 by TommyMurphyTM1234