riscv-collab/riscv-gnu-toolchain #810: Compile a dynamically linked executable

saqibkh, 2020-12-31:
Need some help in generating a dynamically linked executable for c instead of a statically linked executable. I tried using the "-pie" command but that is generating an error.

$ riscv64-unknown-elf-gcc -o quick_sort quick_sort.c
$ file quick_sort
quick_sort: ELF 64-bit LSB executable, UCB RISC-V, version 1 (SYSV), statically linked, with debug_info, not stripped

Thanks

jim-wilson, 2020-12-31:
The embedded elf toolchain does not support dynamic linking or shared libraries.  You need to use a linux toolchain instead.  And note that most simulators won't be able to run a dynamically linked program, you will have to use qemu for that if you aren't already.

saqibkh, 2020-12-31:
Thanks for the explanation.

closed 2020-12-31 by saqibkh