riscv-collab/riscv-gnu-toolchain #812: riscv gdb remote: `info register` shows dscratch and mucounteren

lbmeng, 2021-01-04:
Use gdb 10.1 to do remote debugging with QEMU v5.2.0:

```
>>> info register
ra             0x0      0x0
sp             0x0      0x0
gp             0x0      0x0
tp             0x0      0x0
t0             0x0      0
t1             0x0      0
t2             0x0      0
fp             0x0      0x0
s1             0x0      0
a0             0x0      0
a1             0x0      0
a2             0x0      0
a3             0x0      0
a4             0x0      0
a5             0x0      0
a6             0x0      0
a7             0x0      0
s2             0x0      0
s3             0x0      0
s4             0x0      0
s5             0x0      0
s6             0x0      0
s7             0x0      0
s8             0x0      0
s9             0x0      0
s10            0x0      0
s11            0x0      0
t3             0x0      0
t4             0x0      0
t5             0x0      0
t6             0x0      0
pc             0x1000   0x1000
dscratch       Could not fetch register "dscratch"; remote failure reply 'E14'
mucounteren    Could not fetch register "mucounteren"; remote failure reply 'E14'
```

**dscratch** and **mucounteren** should not be in the register group "general".
Is this a bug of gdb, or QEMU?

jim-wilson, 2021-01-04:
I think gdb is iterating over all registers, trying to find which registers are in the general set.  The target is reporting registers that gdb doesn't know about, so it is reporting errors for them.   I also think that Andrew Burgess has already fixed this in upstream FSF GDB.  IMHO the real problem is that no one is maintaining the qemu gdb stub.  dscratch and mucounteren are priv spec 1.9 registers that binutils/gdb no longer support.  And since qemu no longer supports priv spec 1.9 either it should not be reporting that these registers exist.  See the gdb-xml/riscv-*bit-csr.xml files.  I added the original version of them a while ago, and they haven't been updated since.

For the fix see
    https://sourceware.org/pipermail/gdb-patches/2020-November/173567.html
but not in a gdb release yet.

lbmeng, 2021-01-05:
Thanks Jim. With the following changes on the QEMU side:

```
diff --git a/gdb-xml/riscv-64bit-csr.xml b/gdb-xml/riscv-64bit-csr.xml
index 6aa4bed9f5..fa4cb09d19 100644
--- a/gdb-xml/riscv-64bit-csr.xml
+++ b/gdb-xml/riscv-64bit-csr.xml
@@ -227,7 +227,6 @@
   <reg name="tdata3" bitsize="64"/>
   <reg name="dcsr" bitsize="64"/>
   <reg name="dpc" bitsize="64"/>
-  <reg name="dscratch" bitsize="64"/>
   <reg name="hstatus" bitsize="64"/>
   <reg name="hedeleg" bitsize="64"/>
   <reg name="hideleg" bitsize="64"/>
@@ -244,7 +243,6 @@
   <reg name="mibound" bitsize="64"/>
   <reg name="mdbase" bitsize="64"/>
   <reg name="mdbound" bitsize="64"/>
-  <reg name="mucounteren" bitsize="64"/>
   <reg name="mscounteren" bitsize="64"/>
   <reg name="mhcounteren" bitsize="64"/>
 </feature>
```

The issue is gone with the official GDB 10.1:

```
>>> info registers
ra             0x0      0x0
sp             0x0      0x0
gp             0x0      0x0
tp             0x0      0x0
t0             0x0      0
t1             0x0      0
t2             0x0      0
fp             0x0      0x0
s1             0x0      0
a0             0x0      0
a1             0x0      0
a2             0x0      0
a3             0x0      0
a4             0x0      0
a5             0x0      0
a6             0x0      0
a7             0x0      0
s2             0x0      0
s3             0x0      0
s4             0x0      0
s5             0x0      0
s6             0x0      0
s7             0x0      0
s8             0x0      0
s9             0x0      0
s10            0x0      0
s11            0x0      0
t3             0x0      0
t4             0x0      0
t5             0x0      0
t6             0x0      0
pc             0x1000   0x1000
```

However, I still don't get to the whole issue. You mentioned

> I think gdb is iterating over all registers, trying to find which registers are in the general set.

As you can see in the QEMU riscv-64bit-csr.xml, there is also **mscounteren** and **mhcounteren**. Why aren't there 2 registers not shown in the "info registers"?

> dscratch and mucounteren are priv spec 1.9 registers that binutils/gdb no longer support.

I checked the priv spec 1.9 and found **dscratch** is not listed. **dscratch** is listed in the priv spec 1.10.

jim-wilson, 2021-01-05:
uscounteren and mhcounteren still exist, so gdb knows that they aren't general registers.  dscratch was dropped and replaced with dscratch0 and dscratch1.  mucounteren was dropped and replaced with mcountinhibit.  Anyways, the important issue is that gdb doesn't know what the dscratch and mucounteren registers are anymore, so it gets confused.  Andrew Burgess added a patch to just ignore mysterious unknown registers reported by the target via xml.

The qemu xml files should be updated to have a correct list of csrs that qemu does actually implement.  And this list should depend on the priv spec version, so it might be necessary to create the list at run-tmie, or maybe have multiple lists one for each supported priv-spec version.  As an extra bonus, we should get the csr reg list correct when specifying a target like sifive_u, which depends on the hardware more than the priv spec version, and makes this more complicated.

lbmeng, 2021-01-05:
> uscounteren and mhcounteren still exist, so gdb knows that they aren't general registers. dscratch was dropped and replaced with dscratch0 and dscratch1. mucounteren was dropped and replaced with [...]

Thanks. Now I got it. So it's not a priv 1.9 vs priv 1.10 thing :)

> The qemu xml files should be updated to have a correct list of csrs that qemu does actually implement. And this list should depend on the priv spec version, so it might be necessary to create the [...]

I can try to come up with a QEMU patch to generate a correct list of CSRs that QEMU actually implements. Will send out for you to review then.

closed 2021-01-06 by lbmeng