riscv-collab/riscv-gnu-toolchain #814: How to run target specific tests for GCC?

higuoxing, 2021-01-05:
Hi,

I've followed the [instruction](https://github.com/riscv/riscv-gnu-toolchain#test-suite) to test GCC.

```
./configure --prefix=$RISCV --disable-linux --with-arch=rv64ima # or --with-arch=rv32ima
make newlib
make report-newlib SIM=qemu # I'm using qemu
```

Finally, it prints
```
               ========= Summary of gcc testsuite =========
                            | # of unexpected case / # of unique unexpected case
                            |          gcc |          g++ |     gfortran |
    rv64ima/   lp64/ medlow |    0 /     0 |    0 /     0 |      - |
```

Does it mean that all the tests on the allowlist are passed? I do observed some failures, e.g.,

```
                === g++ Summary ===

# of expected passes            157375
# of unexpected failures        19
# of expected failures          554
# of unsupported tests          8391
/home/v/x/riscv-gnu-toolchain/build-gcc-newlib-stage2/gcc/xg++  version 10.1.0 (GCC) ## I'm using the rvv-intrinsic branch.
```

If all the tests on the allowlist are passed, is there any way to run target specific tests (e.g. RVV related tests)? It takes a very long time to run all the tests.

Thanks in advance!

kito-cheng, 2021-01-05:
Yes, it's possible to run RISC-V specific test only, but you need to modify Makefile manually:
```
diff --git a/Makefile.in b/Makefile.in
index ab2cc05..11064a5 100644
--- a/Makefile.in
+++ b/Makefile.in
@@ -836,7 +836,7 @@ stamps/build-dejagnu: $(srcdir)/riscv-dejagnu $(srcdir)/riscv-dejagnu/.git
        date > $@
 
 stamps/check-gcc-newlib: stamps/build-gcc-newlib-stage2 $(SIM_STAMP) stamps/build-dejagnu
-       $(SIM_PREPARE) $(MAKE) -C build-gcc-newlib-stage2 check-gcc "RUNTESTFLAGS=--target_board='$(NEWLIB_TARGET_BOARDS)'"
+       $(SIM_PREPARE) $(MAKE) -C build-gcc-newlib-stage2 check-gcc "RUNTESTFLAGS=--target_board='$(NEWLIB_TARGET_BOARDS)' riscv.exp riscv-rvv.exp"
        mkdir -p $(dir $@)
        date > $@
 
```

higuoxing, 2021-01-05:
Oh, nice, thank you very much!

closed 2021-01-05 by higuoxing