riscv-collab/riscv-gnu-toolchain #815: riscv newlib: relocation error R_RISCV_GPREL_I against symbol `__preinit_array_start'

LucaZulberti, 2021-01-05:
Hi all,

i'm experiencing this issue when trying to compile an application with newlib toolchain (i'm using latest `master` branch):
```
in function `__libc_init_array': init.c:(.text+0x6): relocation truncated to fit: R_RISCV_GPREL_I against symbol `__preinit_array_start' defined in .init_array section in <elf file>.
```

I use the linker script provided by `riscv64-unknown-elf-ld --verbose` adding this snippet:
```
ENTRY(_pre_start)
MEMORY
{
    /* BRAM memory on FPGA Board */
    ram (rwx)  : ORIGIN = 0x10000000, LENGTH = 128k
}
SECTIONS
{
  /* used by pre_crt0.S to initialize the stack pointer */
  PROVIDE(__stack_top = ORIGIN(ram) + LENGTH(ram));

  /* Read-only sections, merged into text segment: */
  PROVIDE (__executable_start = SEGMENT_START("text-segment", ORIGIN(ram)));
  . = SEGMENT_START("text-segment", ORIGIN(ram)) + SIZEOF_HEADERS;
```

I set the stack pointer in the `_pre_start` function, from where the `_start` function is called:
```
.section .init, "ax"
.global _pre_start
_pre_start:

.option push
.option norelax
	lui	sp,     %hi(__stack_top)
	addi	sp, sp, %lo(__stack_top)

	lui	t0,     %hi(_start)
	addi	t0, t0, %lo(_start)
.option pop
	jr	0(t0)
```

With some applications, I have that link error, I still can't figure out which part of the code is responsible.

In linker script, if I replace:
```
__global_pointer$ = MIN(__SDATA_BEGIN__ + 0x800,
		            MAX(__DATA_BEGIN__ + 0x800, __BSS_END__ - 0x800));
```
with the old:
```
__global_pointer$ = __SDATA_BEGIN__ + 0x800;
```
the relaxation error goes away...

I have no idea why this interfers with section `.init_array` which is at lower addresses than `__DATA_BEGIN__`. Furthermore, the error message says that `__preinit_array_start` symbol is in `.init_array`, but it is actually in `.preinit_array` in the linker script.

Am I missing something?

When the compilation is successful, the applications work correctly on simulation (QuestaSIM) and on FPGA.

Thank you for your support!

LucaZulberti, 2021-01-05:
Thank you for your reply.

I'm using:
```
$ riscv-gnu-toolchain/bin/riscv64-unknown-elf-ld --version
GNU ld (GNU Binutils) 2.35
Copyright (C) 2020 Free Software Foundation, Inc.
This program is free software; you may redistribute it under the terms of
the GNU General Public License version 3 or (at your option) a later version.
This program has absolutely no warranty.
```

The default linker script has got the section .init, and only my function goes there when it compiles. Anyway, declaring it in section .text (just for test), does not remove the error.

I'm going to investigate better in the next days and I will provide more information.
Thank you!

LucaZulberti, 2021-01-05:
Using `-Wl,--noinhibit-exec`, the errors are two:

```
in function `__libc_fini_array':
fini.c:(.text+0x4): relocation truncated to fit: R_RISCV_GPREL_I against symbol `__fini_array_start' defined in .data section in <elf file>

in function `__libc_init_array':
init.c:(.text+0x6): relocation truncated to fit: R_RISCV_GPREL_I against symbol `__preinit_array_start' defined in .init_array section in <elf file>
```

I have provided a custom version (copied from `libgloss`) of `crt0.S` to set the stack pointer, passing `-nostartfiles` to `gcc`. The error persists.

This is the output of `readelf -a`:
```
ELF Header:
  Magic:   7f 45 4c 46 02 01 01 00 00 00 00 00 00 00 00 00 
  Class:                             ELF64
  Data:                              2's complement, little endian
  Version:                           1 (current)
  OS/ABI:                            UNIX - System V
  ABI Version:                       0
  Type:                              EXEC (Executable file)
  Machine:                           RISC-V
  Version:                           0x1
  Entry point address:               0x10000000
  Start of program headers:          64 (bytes into file)
  Start of section headers:          149464 (bytes into file)
  Flags:                             0x1, RVC, soft-float ABI
  Size of this header:               64 (bytes)
  Size of program headers:           56 (bytes)
  Number of program headers:         1
  Size of section headers:           64 (bytes)
  Number of section headers:         21
  Section header string table index: 20

Section Headers:
  [Nr] Name              Type             Address           Offset
       Size              EntSize          Flags  Link  Info  Align
  [ 0]                   NULL             0000000000000000  00000000
       0000000000000000  0000000000000000           0     0     0
  [ 1] .text             PROGBITS         0000000010000000  00001000
       000000000000d8f0  0000000000000000  AX       0     0     2
  [ 2] .rodata           PROGBITS         000000001000d8f0  0000e8f0
       00000000000013b0  0000000000000000   A       0     0     16
  [ 3] .init_array       INIT_ARRAY       000000001000eca0  0000fca0
       0000000000000010  0000000000000008  WA       0     0     8
  [ 4] .data             PROGBITS         000000001000ecb0  0000fcb0
       0000000000001190  0000000000000000  WA       0     0     8
  [ 5] .sdata            PROGBITS         000000001000fe40  00010e40
       0000000000000070  0000000000000000  WA       0     0     8
  [ 6] .sbss             NOBITS           000000001000feb0  00010eb0
       0000000000000038  0000000000000000  WA       0     0     8
  [ 7] .bss              NOBITS           000000001000ff00  00010eb0
       0000000000000160  0000000000000000  WA       0     0     64
  [ 8] .comment          PROGBITS         0000000000000000  00010eb0
       0000000000000012  0000000000000001  MS       0     0     1
  [ 9] .riscv.attributes RISCV_ATTRIBUTE  0000000000000000  00010ec2
       000000000000002b  0000000000000000           0     0     1
  [10] .debug_aranges    PROGBITS         0000000000000000  00010eed
       0000000000000260  0000000000000000           0     0     1
  [11] .debug_info       PROGBITS         0000000000000000  0001114d
       0000000000002b5d  0000000000000000           0     0     1
  [12] .debug_abbrev     PROGBITS         0000000000000000  00013caa
       000000000000113f  0000000000000000           0     0     1
  [13] .debug_line       PROGBITS         0000000000000000  00014de9
       000000000000456c  0000000000000000           0     0     1
  [14] .debug_frame      PROGBITS         0000000000000000  00019358
       00000000000002d0  0000000000000000           0     0     8
  [15] .debug_str        PROGBITS         0000000000000000  00019628
       0000000000000ae2  0000000000000001  MS       0     0     1
[... 72 lines cut ...]
  .preinit_array    :
  {
    PROVIDE_HIDDEN (__preinit_array_start = .);
    KEEP (*(.preinit_array))
    PROVIDE_HIDDEN (__preinit_array_end = .);
  }
  .init_array    :
  {
    PROVIDE_HIDDEN (__init_array_start = .);
    KEEP (*(SORT_BY_INIT_PRIORITY(.init_array.*) SORT_BY_INIT_PRIORITY(.ctors.*)))
    KEEP (*(.init_array EXCLUDE_FILE (*crtbegin.o *crtbegin?.o *crtend.o *crtend?.o ) .ctors))
    PROVIDE_HIDDEN (__init_array_end = .);
  }
  .fini_array    :
  {
    PROVIDE_HIDDEN (__fini_array_start = .);
    KEEP (*(SORT_BY_INIT_PRIORITY(.fini_array.*) SORT_BY_INIT_PRIORITY(.dtors.*)))
    KEEP (*(.fini_array EXCLUDE_FILE (*crtbegin.o *crtbegin?.o *crtend.o *crtend?.o ) .dtors))
    PROVIDE_HIDDEN (__fini_array_end = .);
  }
```

Even if empty `fini_array` and `preinit_array` seems located at 0x1000f000, which is in `.data` section.

Is there an online reference for the ELF standard? The repo `https://github.com/riscv/riscv-elf-psabi-doc` has the paragraph "Sections" empty..

LucaZulberti, 2021-01-08:
Hi @jim-wilson,

I've found something, but I can't explain this behaviour...

Starting removing sections and command from my linker script, I've found that removing the `. = DATA_SEGMENT_END (.);` from it:

```
<...>
  . = SEGMENT_START("ldata-segment", .);
  . = ALIGN(64 / 8);
  __BSS_END__ = .;
    __global_pointer$ = MIN(__SDATA_BEGIN__ + 0x800,
                    MAX(__DATA_BEGIN__ + 0x800, __BSS_END__ - 0x800));
  _end = .; PROVIDE (end = .);
  . = DATA_SEGMENT_END (.);                            <------------
  /* Stabs debugging sections.  */
  .stab          0 : { *(.stab) }
  .stabstr       0 : { *(.stabstr) }
  .stab.excl     0 : { *(.stab.excl) }
<...>
```

resolve the error, and the readelf tool reports:
```
000000001000fc70     0 NOTYPE  LOCAL  DEFAULT    4 __fini_array_end
000000001000fc70     0 NOTYPE  LOCAL  DEFAULT    4 __fini_array_start
000000001000ec80     0 NOTYPE  LOCAL  DEFAULT    3 __init_array_end
000000001000fc70     0 NOTYPE  LOCAL  DEFAULT    3 __preinit_array_end
000000001000ec70     0 NOTYPE  LOCAL  DEFAULT    3 __init_array_start
000000001000fc70     0 NOTYPE  LOCAL  DEFAULT    3 __preinit_array_start
```

Instead of (the only diff between files):
```
000000001000f000     0 NOTYPE  LOCAL  DEFAULT    4 __fini_array_end
000000001000f000     0 NOTYPE  LOCAL  DEFAULT    4 __fini_array_start
000000001000ec80     0 NOTYPE  LOCAL  DEFAULT    3 __init_array_end
000000001000f000     0 NOTYPE  LOCAL  DEFAULT    3 __preinit_array_end
000000001000ec70     0 NOTYPE  LOCAL  DEFAULT    3 __init_array_start
000000001000f000     0 NOTYPE  LOCAL  DEFAULT    3 __preinit_array_start
```

Now the `__libc_init_array` and `__libc_fini_array` function are correctly compiled in:
```
<__libc_init_array>:
    100012f6:	1101                	addi	sp,sp,-32
    100012f8:	e822                	sd	s0,16(sp)
    100012fa:	e04a                	sd	s2,0(sp)
    100012fc:	45018793          	addi	a5,gp,1104 # 1000fc70 <__fini_array_end>
    10001300:	45018913          	addi	s2,gp,1104 # 1000fc70 <__fini_array_end>
    10001304:	40f90933          	sub	s2,s2,a5
    10001308:	ec06                	sd	ra,24(sp)
    1000130a:	e426                	sd	s1,8(sp)
    1000130c:	40395913          	srai	s2,s2,0x3
    10001310:	00090b63          	beqz	s2,10001326 <__libc_init_array+0x30>
    10001314:	45018413          	addi	s0,gp,1104 # 1000fc70 <__fini_array_end>
    10001318:	4481                	li	s1,0
    1000131a:	601c                	ld	a5,0(s0)
    1000131c:	0485                	addi	s1,s1,1
    1000131e:	0421                	addi	s0,s0,8
    10001320:	9782                	jalr	a5
    10001322:	fe991ce3          	bne	s2,s1,1000131a <__libc_init_array+0x24>
    10001326:	1000f437          	lui	s0,0x1000f
    1000132a:	1000f937          	lui	s2,0x1000f
    1000132e:	c7040793          	addi	a5,s0,-912 # 1000ec70 <__init_array_start>
    10001332:	c8090913          	addi	s2,s2,-896 # 1000ec80 <devop_tabs>
    10001336:	40f90933          	sub	s2,s2,a5
    1000133a:	40395913          	srai	s2,s2,0x3
    1000133e:	00090b63          	beqz	s2,10001354 <__libc_init_array+0x5e>
    10001342:	c7040413          	addi	s0,s0,-912
    <...>
```
instead of (due to the linker relocation error):
```
<__libc_init_array>:
    100012f6:	1101                	addi	sp,sp,-32
    100012f8:	e822                	sd	s0,16(sp)
    100012fa:	e04a                	sd	s2,0(sp)
    100012fc:	00040793          	mv	a5,s0
    10001300:	00090913          	mv	s2,s2
    10001304:	40f90933          	sub	s2,s2,a5
    10001308:	ec06                	sd	ra,24(sp)
    1000130a:	e426                	sd	s1,8(sp)
    1000130c:	40395913          	srai	s2,s2,0x3
    10001310:	00090b63          	beqz	s2,10001326 <__libc_init_array+0x30>
    10001314:	00040413          	mv	s0,s0
    10001318:	4481                	li	s1,0
    1000131a:	601c                	ld	a5,0(s0)
    1000131c:	0485                	addi	s1,s1,1
    1000131e:	0421                	addi	s0,s0,8
    10001320:	9782                	jalr	a5
    10001322:	fe991ce3          	bne	s2,s1,1000131a <__libc_init_array+0x24>
    10001326:	00000437          	lui	s0,0x0
    1000132a:	00000937          	lui	s2,0x0
    1000132e:	00040793          	mv	a5,s0
    10001332:	00090913          	mv	s2,s2
    10001336:	40f90933          	sub	s2,s2,a5
    1000133a:	40395913          	srai	s2,s2,0x3
    1000133e:	00090b63          	beqz	s2,10001354 <__libc_init_array+0x5e>
    10001342:	00040413          	mv	s0,s0
    <...>
```

Do you have any advice on how to further investigate the problem?

Thank you for your support.

Nelson1225, 2021-01-11:
It's really hard to find the root cause according to your incomplete information, so we can just guess what had happened.  However, I used to generate the map file by `-M` to see all section and symbol addresses when linking, this is helpful to find the problems.

In general I would suggest not to define MEMORY region and also use `.` to place the output sections at the same time.  But there is only a `ram` region in your script, and the attributes are `rwx`, so I assume all sections will be placed into the `ram` continuously.  I guess the `__global_pointer$` may be set to the unexpected place since the `__BSS_END__ = .;`, when you also define the MEMORY regions.  But even so, linker shouldn't relax the pattern to gp-relative instruction, when the gp isn't near to the data.  Therefore, there must be something that let linker can not estimate the correct gp value, usually caused by the linker script with a very large alignment between output sections, including the DATA_SEGMENT_ALIGN.

If you remove the DATA_SEGMENT_END, then I think you should also remove the DATA_SEGMENT, since they are paired.

Anyway, the linker truncated errors said that your gp is far away from .data and .init_array, but it should be nearly to the data section in the past.  If you just want to move your start address, I would suggest you don't define the MEMORY, just modify the SEGMENT_START address.  Otherwise, you should rewrite your `__global_pointer$` directive, change it to `__global_pointer$ = __SDATA_BEGIN__ + 0x800;` is OK, since it won't refer to the `.`.

Nelson

LucaZulberti, 2021-01-11:
@Nelson1225 Thank you for your precious considerations.

I removed the MEMORY region and everything is fine now. With the MEMORY region defined, there were symbols with a badly estimated address that were not contiguous in memory.

Issue closed.

closed 2021-01-11 by LucaZulberti