riscv-collab/riscv-gnu-toolchain #817: qemu-riscv32: Unable to reserve 0xfffff000 bytes of virtual address space

Dream-math, 2021-01-07:
Hi, when I run
```
qemu-riscv32  cpp_deploy_pack_sys
```
there is an error:
```
qemu-riscv32: Unable to reserve 0xfffff000 bytes of virtual address space (Permission denied) for use as guest address space (check your virtual memory ulimit setting or reserve less using -R option)
```
And the SELinux give an alert:
```
SELinux is preventing qemu-riscv32 from mmap_zero access on the memprotect labeled unconfined_t.
```
Could anyone give some advice? Thanks!

jim-wilson, 2021-01-07:
The rv32 linux support is not as well tested as the rv64 support.  Does it work for rv64 linux?

If you are serious about rv32 linux, then you may need to debug it and fix it yourself.  It could be a glibc, linux kernel, or qemu problem.  if you find the problem you should report it upstream, as no development work is done here.

Dream-math, 2021-01-08:
@jim-wilson , thanks for your reply, it works for rv64 linux, I want to fix it, but I do not know how to do, what I have foud is 
```
(gdb) file ../lib/cpp_deploy_pack_sys 
Reading symbols from ../lib/cpp_deploy_pack_sys...
(gdb) target remote localhost:2000
Remote debugging using localhost:2000
Reading symbols from /home/hye/soft/riscv/new_install_32/sysroot/lib/ld-linux-riscv32-ilp32.so.1...
0x3ffe7ad0 in _start () from /home/hye/soft/riscv/new_install_32/sysroot/lib/ld-linux-riscv32-ilp32.so.1
(gdb) c
Continuing.

Program received signal SIGABRT, Aborted.
__GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
50	  return ret;
```
Does it mean there is something wrong with ld-linux-riscv32-ilp32.so? the program terminated before main function, could you
give me some advice about finding out the problem, Thanks!

jim-wilson, 2021-01-08:
You can set a breakpoint at _start, and try stepping through code to see exactly where it fails.  There are a lot of things that could be wrong here.  There could be an incompatibility between the glibc and qemu as rv32 glibc was upstreamed a few months ago and is not completely compatible with the glibc that is used by riscv-gnu-toolchain.  So you could try older qemu to see if that makes a difference.  Or try using upstream glibc instead of the riscv/riscv-glibc to see if that makes a difference.  Since you have a signal, you can try a gdb backtrace command to see if that gives you any useful info.  There isn't really a specific way to debug this.  You just have to suspect anything might be broken and investigate everything that doesn't work the way you expect.  Trying to cut down your program might help.  Maybe it is a specific system call used by your program that is causing the problem.

Dream-math, 2021-01-08:
@jim-wilson ,thanks for your advice, when I try to use glibc-2.3.1, there is an error
```
checking for egrep... (cached) /usr/bin/grep -E
running configure fragment for sysdeps/unix/sysv/linux
checking installed Linux kernel header files... 3.2.0 or later
checking for kernel header at least 5.10.5... too old!
configure: error: *** The available kernel headers are older than the requested
*** compatible kernel version
make: *** [Makefile:223: stamps/build-glibc-linux-headers] Error 1
```
but my host kernel is 5.10.5, so what's the problem, do I need to replace riscv-gnu-toolchain/linux-headers with
something else? Thanks!

kito-cheng, 2021-01-11:
> but my host kernel is 5.10.5, so what's the problem, do I need to replace riscv-gnu-toolchain/linux-headers with
> something else? Thanks!

Yes, it's 5.0 on the riscv-gnu-toolchain/linux-headers, so you need update or provide your linux header.
You might have question why it not update it, that's because Jim and me don't have time to do that...but we are appreciate if you are interested to testing and upgrade that :)

Dream-math, 2021-01-11:
@kito-cheng thanks for your help, I can have a try after solving my problem, right now, I replace the linux-headers, and compile with riscv-glibc-2.31, my configure is:
```
./configure --prefix=/home/hye/soft/riscv/new_install_32 --with-arch=rv32gc --with-abi=ilp32 --enable-linux
```
the error comes at build-glibc-linux-rv32gc-ilp32:
```
tzfile.c:113:3: error: static assertion failed: "__time64_t must be eight bytes"
  113 |   _Static_assert (sizeof (__time64_t) == 8,
      |   ^~~~~~~~~~~~~~
make[3]: *** [../o-iterator.mk:9: /home/hye/soft/riscv/riscv-gnu-toolchain/build-glibc-linux-rv32gc-ilp32/time/tzfile.o] Error 1
```
I notice that in this page https://github.com/riscv/riscv-glibc/commit/07305b3f41effdba4a1bdcae50af17188aae3fa2, you have
make riscv-glibc-2.31 work on 32 bit rscv, so could you give me some advice to make it work?

kito-cheng, 2021-01-11:
@Dream-math You need using glibc trunk for RV32 support, it not included in glibc 2.32 yet, I've tried last week, verified with gcc testsuite on Qemu, but there is some extra regression failure, I didn't spend too much time yet.

Dream-math, 2021-01-12:
I found where the error comes:
``` C
static __always_inline void
futex_wake (unsigned int *futex_word, int processes_to_wake, int private)
{
  int res = lll_futex_wake (futex_word, processes_to_wake, private);
  /* No error.  Ignore the number of woken processes.  */
  if (res >= 0)
    return;
  switch (res)
    {
    case -EFAULT: /* Could have happened due to memory reuse.  */
    case -EINVAL: /* Could be either due to incorrect alignment (a bug in
                     glibc or in the application) or due to memory being
                     reused for a PI futex.  We cannot distinguish between the
                     two causes, and one of them is correct use, so we do not
                     act in this case.  */
      return;
    case -ENOSYS: /* Must have been caused by a glibc bug.  */
    /* No other errors are documented at this time.  */
    default:
      futex_fatal_error ();
    }
}
```
the res value is -ENOSYS, but I can not understand this code, do I have to learn the mechanism about glibc futex to make my program work, if it is true, I have to suspend this work temporarily, you all help me a lot, thank you very much!

kito-cheng, 2021-01-12:
I don't have time to look that recently, but I guess it could be possible there is unsync between kernel/glibc and qemu for RV32?

kito-cheng, 2021-01-12:
Oh, I just take few more second on your comment, it's seems like qemu issue there, it should not fall in to -ENOSYS.

Dream-math, 2021-01-12:
I just have a last try, I download https://github.com/alistair23/glibc/tree/alistair/rv32.next, compile with linux-kernel-5.10, and it works!

closed 2021-03-03 by lazyparser