riscv-collab/riscv-gnu-toolchain #820: Writing Branch Instructions in Embedded Assembly

daweili1226, 2021-01-13:
When I write code using embedded assembly:

```
__asm __volatile(
    "li       t0, 0x0\n"
    "li       t1, 0x1\n"
    "beq      t0, t1, 0x8\n"
    "li       t0, 0x1\n"
    "sd       t0, 0(%[ptrDest])\n"
    : 
    : [ptrDest] "r" (ptrDest)
    : "t0", "t1", "memory"
    );
```
The compiler cannot generate correct target address for `beq` instruction if I write with an immediate offset 0x8 (as described in RV manual). Then I ran several tests, I found that I should write the target pc address instead of an offset value to the current pc.

```
__asm __volatile(
    "li       t0, 0x0\n"
    "li       t1, 0x1\n"
    "beq      t0, t1, target pc\n"
    "li       t0, 0x1\n"
    "sd       t0, 0(%[ptrDest])\n"
    : 
    : [ptrDest] "r" (ptrDest)
    : "t0", "t1", "memory"
    );
```

Does it mean I can only write branch instructions with labels?

aswaterman, 2021-01-13:
Correct.  The standard approach is to use local labels (see https://sourceware.org/binutils/docs-2.24/as/Symbol-Names.html#Symbol-Names)

This approach also avoids difficulties computing branch offsets in the presence of compressed instructions.  For example, `li t0,1` is compressible, so if RVC is enabled, the branch offset in your example should be 6, not 8.

The ISA manual uses a pseudo-assembly-language format; it's not meant to be a reference for the GNU assembly format.

TommyMurphyTM1234, 2022-09-27:
Question about using GCC inline assembly not an issue with the RISC-V GCC toolchain itself.

closed 2022-09-27 by TommyMurphyTM1234