riscv-collab/riscv-gnu-toolchain #821: Typo in [rvv-intrinsic]?

xicongye, 2021-01-14:
1. commit id:
```
$ cd riscv-gnu-toolchain
$ git branch
  master
  rvv-0.7.x
  rvv-0.8.x
  rvv-0.9.x
* rvv-intrinsic
$ git log -1
commit 1e24571a8c3058a2160f10913e465dfa6f39046d
Merge: 8618434 f6d5752
Author: Kito Cheng <kito.cheng@sifive.com>
Date:   Mon Jan 11 16:35:07 2021 +0800

    Merge pull request #819 from higuoxing/rvv-intrinsic
    
    Specify branch info in .gitmodules.
$ cd riscv-gcc
$ git branch
* (HEAD detached at dc7e524)
  master
  riscv-gcc-8.2.0
$ git log -1
commit dc7e524c23a579120d94912802a6ba174b3a7ed6
Author: Kito Cheng <kito.cheng@sifive.com>
Date:   Thu Sep 3 14:21:19 2020 +0800

    Vector load/store pattern default whole register load/store now
```
2. description:
test code:
```
#include <riscv_vector.h>
#include <stdio.h>      
const int8_t src[16] = {1, 2, 3, 4, 5, 6, 7};
int main(void)
{
    vlseg2e8ff_v_i8m1x2(src);
    return 0;
}
```
compile error:
```
/tmp/ccOBoDvC.s: Assembler messages:
/tmp/ccOBoDvC.s:51: Error: unrecognized opcode `vlseg2eff.v v2,(a4)'
```
3. conclusion
`vlseg2eff.v v2,(a4)` is an illegal instruction, which should be `vlseg2e8ff.v v2,(a4)`. 
It looks like there are some typos in vector.md, maybe the description for vlsegff instructions should be `vlseg<NF>e<sew>ff.v` instead of `vlseg<NF>eff.v` ?

kito-cheng, 2021-01-22:
Thanks for report that! fixed via https://github.com/riscv/riscv-gcc/commit/36be0f0075fb41d73357a79895a214928df075d8

kito-cheng, 2021-01-22:
Just remind you I did't bump gcc for `riscv-gnu-toolchain` yet, could you verified the fix by manually update `riscv-gcc`, and let me know it work for you.

xicongye, 2021-01-27:
I have checkout the `riscv-gcc` you mentioned, and it works fine, thanks!

closed 2021-01-27 by xicongye