riscv-collab/riscv-gnu-toolchain #823: Fail when Using RISCV gcc to build LLVM

drcut, 2021-01-18:
Hi there,
I try to use the gcc/g++ from the toolchain to build LLVM. **Please note, I do not only want to LLVM to generate RISCV file, but also want to LLVM executed on a RISCV environment. This is the difference for my situation with existing llvm-riscv projects**
Here is the command (I have set gcc/g++ in riscv-gnu-toolchain/build for default gcc):
```bash
cmake -G Ninja -DCMAKE_BUILD_TYPE="Debug" \
  -DBUILD_SHARED_LIBS=True -DLLVM_USE_SPLIT_DWARF=True \
  -DLLVM_OPTIMIZED_TABLEGEN=True \
  -DLLVM_BUILD_TESTS=True \
  -DDEFAULT_SYSROOT="/home/riscv-gnu-toolchain/build/riscv32-unknown-elf" \
  -DGCC_INSTALL_PREFIX="/home/riscv-gnu-toolchain/build" \
  -DLLVM_DEFAULT_TARGET_TRIPLE="riscv32-unknown-elf" \
  -DLLVM_EXPERIMENTAL_TARGETS_TO_BUILD="RISCV" ../
```
and the content in CMakeError.log 
```
Determining if the include file malloc/malloc.h exists failed with the following output:
Change Dir: /home/SENSETIME/hanruobing/llvm10/build/CMakeFiles/CMakeTmp

Run Build Command(s):/home/SENSETIME/hanruobing/ninja cmTC_9d5c1 && [1/2] Building C object CMakeFiles/cmTC_9d5c1.dir/CheckIncludeFile.c.o
FAILED: CMakeFiles/cmTC_9d5c1.dir/CheckIncludeFile.c.o 
/home/SENSETIME/hanruobing/riscv64-linux/riscv32-ilp/bin/riscv32-buildroot-linux-gnu-gcc   -I/usr/local/cuda/include -o CMakeFiles/cmTC_9d5c1.dir/CheckIncludeFile.c.o   -c CheckIncludeFile.c
CheckIncludeFile.c:1:10: fatal error: malloc/malloc.h: No such file or directory
    1 | #include <malloc/malloc.h>
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
ninja: build stopped: subcommand failed.

Determining if the include file valgrind/valgrind.h exists failed with the following output:
Change Dir: /home/SENSETIME/hanruobing/llvm10/build/CMakeFiles/CMakeTmp

Run Build Command(s):/home/SENSETIME/hanruobing/ninja cmTC_c9f68 && [1/2] Building C object CMakeFiles/cmTC_c9f68.dir/CheckIncludeFile.c.o
FAILED: CMakeFiles/cmTC_c9f68.dir/CheckIncludeFile.c.o 
/home/SENSETIME/hanruobing/riscv64-linux/riscv32-ilp/bin/riscv32-buildroot-linux-gnu-gcc   -I/usr/local/cuda/include -o CMakeFiles/cmTC_c9f68.dir/CheckIncludeFile.c.o   -c CheckIncludeFile.c
CheckIncludeFile.c:1:10: fatal error: valgrind/valgrind.h: No such file or directory
    1 | #include <valgrind/valgrind.h>
      |          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
ninja: build stopped: subcommand failed.

Determining if the include file zlib.h exists failed with the following output:
Change Dir: /home/SENSETIME/hanruobing/llvm10/build/CMakeFiles/CMakeTmp

Run Build Command(s):/home/SENSETIME/hanruobing/ninja cmTC_e83c3 && [1/2] Building C object CMakeFiles/cmTC_e83c3.dir/CheckIncludeFile.c.o
FAILED: CMakeFiles/cmTC_e83c3.dir/CheckIncludeFile.c.o 
/home/SENSETIME/hanruobing/riscv64-linux/riscv32-ilp/bin/riscv32-buildroot-linux-gnu-gcc   -I/usr/local/cuda/include -o CMakeFiles/cmTC_e83c3.dir/CheckIncludeFile.c.o   -c CheckIncludeFile.c
CheckIncludeFile.c:1:10: fatal error: zlib.h: No such file or directory
    1 | #include <zlib.h>
      |          ^~~~~~~~
compilation terminated.
ninja: build stopped: subcommand failed.

Determining if the include file mach/mach.h exists failed with the following output:
Change Dir: /home/SENSETIME/hanruobing/llvm10/build/CMakeFiles/CMakeTmp

Run Build Command(s):/home/SENSETIME/hanruobing/ninja cmTC_174ad && [1/2] Building C object CMakeFiles/cmTC_174ad.dir/CheckIncludeFile.c.o
FAILED: CMakeFiles/cmTC_174ad.dir/CheckIncludeFile.c.o 
/home/SENSETIME/hanruobing/riscv64-linux/riscv32-ilp/bin/riscv32-buildroot-linux-gnu-gcc   -I/usr/local/cuda/include -o CMakeFiles/cmTC_174ad.dir/CheckIncludeFile.c.o   -c CheckIncludeFile.c
CheckIncludeFile.c:1:10: fatal error: mach/mach.h: No such file or directory
    1 | #include <mach/mach.h>
      |          ^~~~~~~~~~~~~
compilation terminated.
ninja: build stopped: subcommand failed.

Determining if the include file histedit.h exists failed with the following output:
Change Dir: /home/SENSETIME/hanruobing/llvm10/build/CMakeFiles/CMakeTmp

Run Build Command(s):/home/SENSETIME/hanruobing/ninja cmTC_80f79 && [1/2] Building C object CMakeFiles/cmTC_80f79.dir/CheckIncludeFile.c.o
FAILED: CMakeFiles/cmTC_80f79.dir/CheckIncludeFile.c.o 
/home/SENSETIME/hanruobing/riscv64-linux/riscv32-ilp/bin/riscv32-buildroot-linux-gnu-gcc   -I/usr/local/cuda/include -o CMakeFiles/cmTC_80f79.dir/CheckIncludeFile.c.o   -c CheckIncludeFile.c
CheckIncludeFile.c:1:10: fatal error: histedit.h: No such file or directory
[... 483 lines cut ...]
Change Dir: /home/SENSETIME/hanruobing/llvm10/build/CMakeFiles/CMakeTmp

Run Build Command(s):/home/SENSETIME/hanruobing/ninja cmTC_a96e0 && [1/2] Building CXX object CMakeFiles/cmTC_a96e0.dir/src.cxx.o
FAILED: CMakeFiles/cmTC_a96e0.dir/src.cxx.o 
/home/SENSETIME/hanruobing/riscv64-linux/riscv32-ilp/bin/riscv32-buildroot-linux-gnu-g++    -fPIC -fvisibility-inlines-hidden -Werror=date-time -Wall -Wextra -Wno-unused-parameter -Wwrite-strings -Wcast-qual -Wno-missing-field-initializers -pedantic -Wno-long-long -Wimplicit-fallthrough -Wno-maybe-uninitialized -Wno-class-memaccess -Wno-redundant-move -Wno-noexcept-type -Wdelete-non-virtual-dtor -Wno-comment -fdiagnostics-color  -std=c++11  -Wall  -Wextra  -Wshadow  -pedantic  -pedantic-errors  -Wfloat-equal  -fstrict-aliasing  -fno-exceptions  -Wstrict-aliasing -DHAVE_CXX_FLAG_WD654  -wd654   -wd654 -o CMakeFiles/cmTC_a96e0.dir/src.cxx.o -c src.cxx
riscv32-buildroot-linux-gnu-g++.br_real: error: unrecognized command line option '-wd654'
riscv32-buildroot-linux-gnu-g++.br_real: error: unrecognized command line option '-wd654'
ninja: build stopped: subcommand failed.

Source file was:
int main() { return 0; }
Performing C++ SOURCE FILE Test HAVE_CXX_FLAG_WTHREAD_SAFETY failed with the following output:
Change Dir: /home/SENSETIME/hanruobing/llvm10/build/CMakeFiles/CMakeTmp

Run Build Command(s):/home/SENSETIME/hanruobing/ninja cmTC_1d748 && [1/2] Building CXX object CMakeFiles/cmTC_1d748.dir/src.cxx.o
FAILED: CMakeFiles/cmTC_1d748.dir/src.cxx.o 
/home/SENSETIME/hanruobing/riscv64-linux/riscv32-ilp/bin/riscv32-buildroot-linux-gnu-g++    -fPIC -fvisibility-inlines-hidden -Werror=date-time -Wall -Wextra -Wno-unused-parameter -Wwrite-strings -Wcast-qual -Wno-missing-field-initializers -pedantic -Wno-long-long -Wimplicit-fallthrough -Wno-maybe-uninitialized -Wno-class-memaccess -Wno-redundant-move -Wno-noexcept-type -Wdelete-non-virtual-dtor -Wno-comment -fdiagnostics-color  -std=c++11  -Wall  -Wextra  -Wshadow  -pedantic  -pedantic-errors  -Wfloat-equal  -fstrict-aliasing  -fno-exceptions  -Wstrict-aliasing -DHAVE_CXX_FLAG_WTHREAD_SAFETY  -Wthread-safety   -Wthread-safety -o CMakeFiles/cmTC_1d748.dir/src.cxx.o -c src.cxx
riscv32-buildroot-linux-gnu-g++.br_real: error: unrecognized command line option '-Wthread-safety'
riscv32-buildroot-linux-gnu-g++.br_real: error: unrecognized command line option '-Wthread-safety'
ninja: build stopped: subcommand failed.

Source file was:
int main() { return 0; }
```
Does anyone have any suggestions? Thanks

jim-wilson, 2021-01-29:
Cross building a compiler to run on the target is complicated.  You can't use riscv-gnu-toolchain to do this.  You will need to do it the usual FSF or distro way, which is far too complicated to explain in detail here.  You really shouldn't try this unless you have a lot of time available to learn how to cross build compilers.

The simplest solution is to use a distro that already supports llvm.  Fedora does, and Debian probably also.  Just boot them, install the llvm package, and you are done.

If you want a distro that you can build from scratch that supports llvm, then you can do that with openembedded.  The SiFive version of OpenEmbedded (sifive/freedom-u-sdk) is already set up to build a native llvm compiler to run on the target.

Otherwise, you should boot your distro, install gcc, and then use it to build llvm natively, the usual way.

If your distro doesn't already have a native gcc, then you should probably switch distros.  It looks like you are using buildroot.  A native llvm on buildroot is probably not feasible.  Buildroot probably doesn't support enough of a linux system to support a native llvm.

If you really want to do this the hard way, then the best approach, assuming that you already have a target distro, is to make a sysroot from the target root filesystem, and then configure and build the compiler using this sysroot.  This works with gcc.  I don't know if it works with LLVM.  This requires building two compilers.  The first one is build cross to target, e.g. x86_64-linux cross to riscv64-linux.  Then the second one is build=x86_64-linux host=riscv64-linux target=riscv64-linux which will use the first cross compiler to generate the riscv64-linux native code.  A sysroot contains everything the compiler needs, e.g. /usr/include, /usr/lib, /lib, and 64-bit variations if they exist.

If you don't already have a target root filesystem, then it gets a lot harder, too hard to describe here.

TommyMurphyTM1234, 2022-10-10:
Based on @jim-wilson's previous comment I'm closing this on the basis that it's not a valid issue.

closed 2022-10-10 by TommyMurphyTM1234