riscv-collab/riscv-gnu-toolchain #824: m1 mac

xhw1, 2021-01-18:
Hi, 

Is there a way to install `riscv-gnu-toolchain` on m1 Macs?
when I run `brew install riscv-tools` I got error:

```
==> Installing riscv/riscv/riscv-tools dependency: riscv-gnu-toolchain
		==> ./configure --prefix=/opt/homebrew/Cellar/riscv-gnu-toolchain/master
==> make
Last 15 lines from /Users/xiaohan/Library/Logs/Homebrew/riscv-gnu-toolchain/02.make:
      toplev::main(int, char**) in libbackend.a(toplev.o)
      gt_pch_save(__sFILE*) in libbackend.a(ggc-common.o)
      gt_pch_save(__sFILE*) in libbackend.a(ggc-common.o)
      gt_pch_restore(__sFILE*) in libbackend.a(ggc-common.o)
      gt_pch_restore(__sFILE*) in libbackend.a(ggc-common.o)
ld: symbol(s) not found for architecture arm64
ld: symbol(s) not found for architecture arm64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [cc1] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: *** [cc1plus] Error 1
rm gcov.pod fsf-funding.pod lto-dump.pod gpl.pod cpp.pod gfdl.pod gcc.pod gcov-dump.pod gcov-tool.pod
make[1]: *** [all-gcc] Error 2
make: *** [stamps/build-gcc-newlib-stage1] Error 2
```

kito-cheng, 2021-01-19:
Seems like same issue as https://github.com/riscv/riscv-gnu-toolchain/issues/800 , but I am working on other stuff now and I  don't have mac machine to verify that...

xhw1, 2021-01-19:
thanks for the response! I'll close this issue and watch that one

closed 2021-01-19 by xhw1