riscv-collab/riscv-gnu-toolchain #828: Configure script fails with gcc 10.2.1 under Debian cid

barbem, 2021-01-21:
Hi,
I'm working on Debian sid and trying to compile the riscv-gnu-toolchain super project.
The current version of gcc doesn't seem to be supported by the configuration script.

```
$> ./configure
checking for gcc... gcc
checking whether the C compiler works... no
configure: error: in `/home/raifer/ensi/riscv-gnu-toolchain':
configure: error: C compiler cannot create executables
See `config.log' for more details

$> cat config.log
This file contains any messages produced by compilers while
running configure, to aid debugging if configure makes a mistake.

It was created by riscv-toolchain configure 1.0, which was
generated by GNU Autoconf 2.69.  Invocation command line was

  $ ./configure 

## --------- ##
## Platform. ##
## --------- ##

hostname = X1
uname -m = x86_64
uname -r = 4.19.128-microsoft-standard
uname -s = Linux
uname -v = #1 SMP Tue Jun 23 12:58:10 UTC 2020

/usr/bin/uname -p = unknown
/bin/uname -X     = unknown

/bin/arch              = unknown
/usr/bin/arch -k       = unknown
/usr/convex/getsysinfo = unknown
/usr/bin/hostinfo      = unknown
/bin/machine           = unknown
/usr/bin/oslevel       = unknown
/bin/universe          = unknown

PATH: /mnt/c/Windows
PATH: /qemu-cep/bin
PATH: /sbin
PATH: /usr/local/bin
PATH: /usr/bin
PATH: /bin
PATH: /usr/local/games
PATH: /usr/games

## ----------- ##
## Core tests. ##
## ----------- ##

configure:1932: checking for gcc
configure:1948: found /usr/bin/gcc
configure:1959: result: gcc
configure:2188: checking for C compiler version
configure:2197: gcc --version >&5
gcc (Debian 10.2.1-6) 10.2.1 20210110
Copyright (C) 2020 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

configure:2208: $? = 0
configure:2197: gcc -v >&5
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/10/lto-wrapper
OFFLOAD_TARGET_NAMES=nvptx-none:amdgcn-amdhsa:hsa
OFFLOAD_TARGET_DEFAULT=1
Target: x86_64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Debian 10.2.1-6' --with-bugurl=file:///usr/share/doc/gcc-10/README.Bugs --enable-languages=c,ada,c++,go,brig,d,fortran,objc,obj-c++,m2 --prefix=/usr --with-gcc-major-version-only --program-suffix=-10 --program-prefix=x86_64-linux-gnu- --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --enable-bootstrap --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-vtable-verify --enable-plugin --enable-default-pie --with-system-zlib --enable-libphobos-checking=release --with-target-system-zlib=auto --enable-objc-gc=auto --enable-multiarch --disable-werror --with-arch-32=i686 --with-abi=m64 --with-multilib-list=m32,m64,mx32 --enable-multilib --with-tune=generic --enable-offload-targets=nvptx-none=/build/gcc-10-Km9U7s/gcc-10-10.2.1/debian/tmp-nvptx/usr,amdgcn-amdhsa=/build/gcc-10-Km9U7s/gcc-10-10.2.1/debian/tmp-gcn/usr,hsa --without-cuda-driver --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu --with-build-config=bootstrap-lto-lean --enable-link-mutex
Thread model: posix
Supported LTO compression algorithms: zlib zstd
gcc version 10.2.1 20210110 (Debian 10.2.1-6) 
... rest of stderr output deleted ...
configure:2208: $? = 0
configure:2197: gcc -V >&5
gcc: error: unrecognized command-line option '-V'
gcc: fatal error: no input files
compilation terminated.
configure:2208: $? = 1
configure:2197: gcc -qversion >&5
gcc: error: unrecognized command-line option '-qversion'; did you mean '--version'?
gcc: fatal error: no input files
compilation terminated.
configure:2208: $? = 1
configure:2228: checking whether the C compiler works
configure:2250: gcc    conftest.c  >&5
gcc: fatal error: cannot execute 'as': execvp: Input/output error
compilation terminated.
configure:2254: $? = 1
configure:2292: result: no
configure: failed program was:
| /* confdefs.h */
| #define PACKAGE_NAME "riscv-toolchain"
| #define PACKAGE_TARNAME "riscv-toolchain"
| #define PACKAGE_VERSION "1.0"
| #define PACKAGE_STRING "riscv-toolchain 1.0"
| #define PACKAGE_BUGREPORT ""
| #define PACKAGE_URL ""
| /* end confdefs.h.  */
| 
| int
| main ()
| {
| 
|   ;
|   return 0;
| }
configure:2297: error: in `/home/raifer/ensi/riscv-gnu-toolchain':
configure:2299: error: C compiler cannot create executables
See `config.log' for more details

## ---------------- ##
## Cache variables. ##
## ---------------- ##

ac_cv_env_CC_set=
ac_cv_env_CC_value=
ac_cv_env_CFLAGS_set=
ac_cv_env_CFLAGS_value=
ac_cv_env_CPPFLAGS_set=
ac_cv_env_CPPFLAGS_value=
ac_cv_env_LDFLAGS_set=
ac_cv_env_LDFLAGS_value=
ac_cv_env_LIBS_set=
ac_cv_env_LIBS_value=
ac_cv_env_build_alias_set=
ac_cv_env_build_alias_value=
ac_cv_env_host_alias_set=
ac_cv_env_host_alias_value=
ac_cv_env_target_alias_set=
ac_cv_env_target_alias_value=
ac_cv_prog_ac_ct_CC=gcc

## ----------------- ##
## Output variables. ##
## ----------------- ##

BASH='/bin/bash'
CC='gcc'
CFLAGS=''
CPPFLAGS=''
CURL=''
DEFS=''
ECHO_C=''
ECHO_N='-n'
ECHO_T=''
EXEEXT=''
FETCHER=''
FGREP=''
FTP=''
GAWK=''
GREP=''
GSED=''
LDFLAGS=''
LIBOBJS=''
LIBS=''
LTLIBOBJS=''
NEED_GCC_EXTERNAL_LIBRARIES=''
OBJEXT=''
PACKAGE_BUGREPORT=''
PACKAGE_NAME='riscv-toolchain'
PACKAGE_STRING='riscv-toolchain 1.0'
PACKAGE_TARNAME='riscv-toolchain'
PACKAGE_URL=''
PACKAGE_VERSION='1.0'
PATH_SEPARATOR=':'
SHELL='/bin/bash'
WGET=''
WITH_ABI=''
WITH_ARCH=''
WITH_TUNE=''
ac_ct_CC='gcc'
bindir='${exec_prefix}/bin'
build_alias=''
cmodel=''
configure_host=''
datadir='${datarootdir}'
datarootdir='${prefix}/share'
default_target=''
docdir='${datarootdir}/doc/${PACKAGE_TARNAME}'
dvidir='${docdir}'
enable_gdb=''
exec_prefix='NONE'
gcc_checking=''
glibc_multilib_names=''
host_alias=''
htmldir='${docdir}'
includedir='${prefix}/include'
infodir='${datarootdir}/info'
libdir='${exec_prefix}/lib'
libexecdir='${exec_prefix}/libexec'
localedir='${datarootdir}/locale'
localstatedir='${prefix}/var'
mandir='${datarootdir}/man'
multilib_flags=''
multilib_gen=''
newlib_multilib_names=''
oldincludedir='/usr/include'
pdfdir='${docdir}'
prefix='NONE'
program_transform_name='s,x,x,'
psdir='${docdir}'
runstatedir='${localstatedir}/run'
sbindir='${exec_prefix}/sbin'
sharedstatedir='${prefix}/com'
sysconfdir='${prefix}/etc'
target_alias=''
target_cflags=''
target_cxxflags=''
with_binutils_src=''
with_gcc_src=''
with_gdb_src=''
with_glibc_src=''
with_guile=''
with_linux_headers_src=''
with_musl_src=''
with_newlib_src=''
with_system_zlib=''

## ----------- ##
## confdefs.h. ##
## ----------- ##

/* confdefs.h */
#define PACKAGE_NAME "riscv-toolchain"
#define PACKAGE_TARNAME "riscv-toolchain"
#define PACKAGE_VERSION "1.0"
#define PACKAGE_STRING "riscv-toolchain 1.0"
#define PACKAGE_BUGREPORT ""
#define PACKAGE_URL ""

configure: exit 77

$> gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/10/lto-wrapper
OFFLOAD_TARGET_NAMES=nvptx-none:amdgcn-amdhsa:hsa
OFFLOAD_TARGET_DEFAULT=1
Target: x86_64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Debian 10.2.1-6' --with-bugurl=file:///usr/share/doc/gcc-10/README.Bugs --enable-languages=c,ada,c++,go,brig,d,fortr
an,objc,obj-c++,m2 --prefix=/usr --with-gcc-major-version-only --program-suffix=-10 --program-prefix=x86_64-linux-gnu- --enable-shared --enable-linker-build-id --libexecdir
=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --enable-bootstrap --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdc
xx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-vtable-verify --enable-plugin --enable-default-pie --with-system-zlib --enable-libphobos-c
hecking=release --with-target-system-zlib=auto --enable-objc-gc=auto --enable-multiarch --disable-werror --with-arch-32=i686 --with-abi=m64 --with-multilib-list=m32,m64,mx3
2 --enable-multilib --with-tune=generic --enable-offload-targets=nvptx-none=/build/gcc-10-Km9U7s/gcc-10-10.2.1/debian/tmp-nvptx/usr,amdgcn-amdhsa=/build/gcc-10-Km9U7s/gcc-1
0-10.2.1/debian/tmp-gcn/usr,hsa --without-cuda-driver --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu --with-build-conf
ig=bootstrap-lto-lean --enable-link-mutex
Thread model: posix
Supported LTO compression algorithms: zlib zstd
gcc version 10.2.1 20210110 (Debian 10.2.1-6)

$> gcc -V
gcc: error: unrecognized command-line option ‘-V’
gcc: fatal error: no input files
compilation terminated.
`

kito-cheng, 2021-01-22:
The root cause is seems like your gcc can't compile program due to the assembler (as) not work as expected? but it's `Input/output error`, so maybe you remove whole build folder and try again?
```
configure:2250: gcc    conftest.c  >&5
gcc: fatal error: cannot execute 'as': execvp: Input/output error
compilation terminated.
configure:2254: $? = 1
configure:2292: result: no
configure: failed program was:
| /* confdefs.h */
| #define PACKAGE_NAME "riscv-toolchain"
| #define PACKAGE_TARNAME "riscv-toolchain"
| #define PACKAGE_VERSION "1.0"
| #define PACKAGE_STRING "riscv-toolchain 1.0"
| #define PACKAGE_BUGREPORT ""
| #define PACKAGE_URL ""
| /* end confdefs.h.  */
| 
| int
| main ()
| {
| 
|   ;
|   return 0;
| }
configure:2297: error: in `/home/raifer/ensi/riscv-gnu-toolchain':
configure:2299: error: C compiler cannot create executables
See `config.log' for more details
```

barbem, 2021-01-24:
Hi,
thank you for the your answer. Effectively, my as program does not work with a simple C main function.
I will investigate the problem and close this issue.

closed 2021-01-24 by barbem

barbem, 2021-01-24:
it was my Microsoft WSL2 container that no longer worked. A simple restart of Windows and the problem was solved.
Thank you for the help.