riscv-collab/riscv-gnu-toolchain #829: Soft-float using -march=rv32i -mabi=ilp32

waleedbinehsan-lm, 2021-01-22:
Hey I have build the tool-chain using --enable-multilib with installation path /opt/riscv/bin
Now when i run the code using the following command:

/opt/riscv/bin/riscv64-unknown-linux-gnu-gcc /home/waleedbinehsan/Desktop/main.c -lm -lgcc -march=rv32i -mabi=ilp32 -o main

I get the error:
target emulation `elf64-littleriscv' does not match `elf32-littleriscv'

What an I doing wrong? Thanks for your answer in advance

kito-cheng, 2021-01-25:
Hmmmm, could you paste result of `/opt/riscv/bin/riscv64-unknown-elf-gcc -v`?

waleedbinehsan-lm, 2021-01-25:
Using built-in specs.
COLLECT_GCC=/home/waleedbinehsan/chipyard/riscv-tools-install/bin/riscv64-unknown-elf-gcc
Target: riscv64-unknown-elf
Configured with: /home/waleedbinehsan/chipyard/toolchains/riscv-tools/riscv-gnu-toolchain/build/../riscv-gcc/configure --target=riscv64-unknown-elf --prefix=/home/waleedbinehsan/chipyard/riscv-tools-install --disable-shared --disable-threads --enable-languages=c,c++ --with-system-zlib --enable-tls --with-newlib --with-sysroot=/home/waleedbinehsan/chipyard/riscv-tools-install/riscv64-unknown-elf --with-native-system-header-dir=/include --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libgomp --disable-nls --disable-tm-clone-registry --src=../../riscv-gcc --enable-checking=yes --disable-multilib --with-abi=lp64d --with-arch=rv64imafdc --with-tune=rocket 'CFLAGS_FOR_TARGET=-Os  -mcmodel=medany' 'CXXFLAGS_FOR_TARGET=-Os  -mcmodel=medany'
Thread model: single
gcc version 9.2.0 (GCC) 
COLLECT_GCC_OPTIONS='-v' '-march=rv32i' '-mabi=ilp32' '-o' '-' '-S' '-O3' '-mtune=rocket'
 /home/waleedbinehsan/chipyard/riscv-tools-install/libexec/gcc/riscv64-unknown-elf/9.2.0/cc1 -quiet -v /home/waleedbinehsan/Desktop/main.c -quiet -dumpbase main.c -march=rv32i -mabi=ilp32 -mtune=rocket -auxbase-strip - -O3 -version -o -
GNU C17 (GCC) version 9.2.0 (riscv64-unknown-elf)
	compiled by GNU C version 7.5.0, GMP version 6.1.2, MPFR version 4.0.1, MPC version 1.1.0, isl version none
warning: GMP header version 6.1.2 differs from library version 6.2.0.
warning: MPFR header version 4.0.1 differs from library version 4.0.2.
GGC heuristics: --param ggc-min-expand=30 --param ggc-min-heapsize=4096
ignoring nonexistent directory "/home/waleedbinehsan/chipyard/riscv-tools-install/riscv64-unknown-elf/usr/local/include"
ignoring duplicate directory "/home/waleedbinehsan/chipyard/riscv-tools-install/riscv64-unknown-elf/include"
#include "..." search starts here:
#include <...> search starts here:
 /home/waleedbinehsan/chipyard/riscv-tools-install/lib/gcc/riscv64-unknown-elf/9.2.0/include
 /home/waleedbinehsan/chipyard/riscv-tools-install/lib/gcc/riscv64-unknown-elf/9.2.0/include-fixed
 /home/waleedbinehsan/chipyard/riscv-tools-install/lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/include
End of search list.
GNU C17 (GCC) version 9.2.0 (riscv64-unknown-elf)
	compiled by GNU C version 7.5.0, GMP version 6.1.2, MPFR version 4.0.1, MPC version 1.1.0, isl version none
warning: GMP header version 6.1.2 differs from library version 6.2.0.
warning: MPFR header version 4.0.1 differs from library version 4.0.2.
GGC heuristics: --param ggc-min-expand=30 --param ggc-min-heapsize=4096
Compiler executable checksum: 409deaeed954ac3a473c843219e0bf69
COMPILER_PATH=/home/waleedbinehsan/chipyard/riscv-tools-install/libexec/gcc/riscv64-unknown-elf/9.2.0/:/home/waleedbinehsan/chipyard/riscv-tools-install/libexec/gcc/riscv64-unknown-elf/9.2.0/:/home/waleedbinehsan/chipyard/riscv-tools-install/libexec/gcc/riscv64-unknown-elf/:/home/waleedbinehsan/chipyard/riscv-tools-install/lib/gcc/riscv64-unknown-elf/9.2.0/:/home/waleedbinehsan/chipyard/riscv-tools-install/lib/gcc/riscv64-unknown-elf/:/home/waleedbinehsan/chipyard/riscv-tools-install/lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/bin/
LIBRARY_PATH=/home/waleedbinehsan/chipyard/riscv-tools-install/lib/gcc/riscv64-unknown-elf/9.2.0/:/home/waleedbinehsan/chipyard/riscv-tools-install/lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/lib/:/home/waleedbinehsan/chipyard/riscv-tools-install/riscv64-unknown-elf/lib/
COLLECT_GCC_OPTIONS='-v' '-march=rv32i' '-mabi=ilp32' '-o' '-' '-S' '-O3' '-mtune=rocket'

These are the results when i run GCC with -v flag

kito-cheng, 2021-01-25:
GCC has been upgraded to 10.2 at 2020.05,  and I can't reproduce your issue with the current source tree, so I would suggest your upgrade your toolchain and try again?

waleedbinehsan-lm, 2021-01-26:
I have downloaded and installed a newer tool-chain. By using the following commands

./configure --prefix=/opt/riscv --enable-multilib
sudo make

After that i ran the
/opt/riscv/bin/riscv64-unknown-elf-gcc /home/waleedbinehsan/Desktop/main.c -lm -lgcc -march=rv32i -mabi=ilp32 -o main

Yet still its calling the __adddf3 routines how can i get the asm file that have been linked the respective libraries call

kito-cheng, 2021-01-27:
@waleedbinehsan-lm , so the issue `target emulation elf64-littleriscv' does not match elf32-littleriscv'` is gone after install new toolchain?

> Yet still its calling the __adddf3 routines how can i get the asm file that have been linked the respective libraries call

Are you intend to get a binary without adddf3 function call when build program with `-march=rv32i -mabi=ilp32`? It's almost impossible unless you don't use libc from newlib.

waleedbinehsan-lm, 2021-01-27:
Hey yes the 
''target emulation elf64-littleriscv' does not match elf32-littleriscv''' is fixed.

I actually want to run the code on a core, so I made object file using the following command
''/opt/riscv/bin/riscv64-unknown-elf-gcc -I SweRV-Original/SweRV-EL2_older_release/snapshots/default/ -g -O3 -funroll-all-loops -mabi=ilp32 -march=rv32imc -nostdlib -c main.c -o main.o''

After that to make the exe file i used the following command
''/opt/riscv/bin/riscv64-unknown-elf-ld -m elf32lriscv --discard-none -T SweRV-Original/SweRV-EL2_older_release/testbench/link.ld -o main.exe main.o''

But I am now getting errors such as:
/opt/riscv/bin/riscv64-unknown-elf-ld: /home/waleedbinehsan/Desktop/main.c:118: undefined reference to `__adddf3'

kito-cheng, 2021-01-27:
Add `-lgcc`  to your LDFLAGS should fix the issue.

waleedbinehsan-lm, 2021-01-27:
Still having the same issue, I am using the following commands:
To generate the object file
/opt/riscv/bin/riscv64-unknown-elf-gcc -I SweRV-Original/SweRV-EL2_older_release/snapshots/default/ -g -O3 -funroll-all-loops **-lgcc** -mabi=ilp32 -march=rv32imc -c main.c -o main.o       

To generate exe file
/opt/riscv/bin/riscv64-unknown-elf-ld -m elf32lriscv --discard-none -T SweRV-Original/SweRV-EL2_older_release/testbench/link.ld -o main.exe main.o

waleedbinehsan-lm, 2021-01-27:
I also tried
Still having the same issue, I am using the following commands:
To generate the object file
/opt/riscv/bin/riscv64-unknown-elf-gcc -I SweRV-Original/SweRV-EL2_older_release/snapshots/default/ -g -O3 -funroll-all-loops **-lgcc -lm** -mabi=ilp32 -march=rv32imc -c main.c -o main.o

To generate exe file
/opt/riscv/bin/riscv64-unknown-elf-ld -m elf32lriscv **-lgcc -lm** --discard-none -T SweRV-Original/SweRV-EL2_older_release/testbench/link.ld -o main.exe main.o

Still the same issue
/opt/riscv/bin/riscv64-unknown-elf-ld: /home/waleedbinehsan/Desktop/main.c:118: undefined reference to `__adddf3'

waleedbinehsan-lm, 2021-01-29:
Thank you very much i was able to resolve the problem, by using -gcc to compile and link -lm -lgcc and made .exe from that.

closed 2021-01-29 by waleedbinehsan-lm

HamzaShabbir517, 2022-08-21:
@waleedbinehsan-personal can u please tell me what gcc command you use to solve the above issue as i am facing the same issue. I am using SweRV-EL2 default makefile and i try the same things as mention in this issue but it seem i have no luck.