riscv-collab/riscv-gnu-toolchain #831: Help with the cross-compiler for a newbie

Preetderasari14, 2021-01-26:
Hi! I recently downloaded and installed the Linux cross-compiler (without multilib support i.e. only 64 bit). I was looking for some tips on using the toolchain vis-a-vis how to compile a basic c or cpp code, understanding some options I might need to use while compiling a more complicated code, and if I require multilib support in the future do I uninstall the current version of the toolchain? (if so how?) or do I just make the code again and it will automatically reinstall the toolchain. Also, I would like to know if I installed the right cross-compiler (i.e. if Newlib/elf is a better option). 

P.S. I mainly want to use my compiled binaries with the gem5 simulator, so any tips with that would also be appreciated. 

Thank you.

OliverrrrWu, 2021-01-27:
morning.
As you said， you have installed the 'riscv-gnu-toolchain', right? that means you probably have run the command like `./configure --prefix=/opt/riscv`. the arguments `--prefix=/opt/riscv` means the toolchain will be installed into the folder under this path. if you want to uninstall it, you just remove this folder. 
if you want to reinstall, may you just re-run the configure program with your arguments like `./configure --prefix=/opt/riscv --enable-multilib`, if you want to cover the original toolchain, please make sure the arguments `--prefix` is same as before. then run the command `make` or `make linux` (it depends on what you want).
To check you have installed the right cross-compiler, find the folder where you install them. if it is `/opt/riscv/`, run command `/opt/riscv/riscv64-unknown-elf-gcc -v` if you have install the Newlib/elf. if not, run `/opt/riscv/riscv64-unknown-linux-gnu-gcc -v`.
wish it could help you.
if you are able to read Chinese, this link may help you.[How to build a RISC-V environment](https://www.wulongxin.com/index.php/archives/20210110908.html)

Preetderasari14, 2021-02-01:
Hi @OliverrrrWu thank you for your reply. I did exactly as you mentioned. I deleted the existing toolchain using `sudo rm -r /opt/riscv` and I think that worked. Then I did a fresh installation of the Newlib version of the toolchain using the multilib command (since I am not sure what bit instructions I need). The `sudo make` command also ran perfectly and it built the toolchain. But when I try to do '/opt/riscv/riscv64-unknown-linux-gnu-gcc -V` it gives me error saying `bash: /opt/riscv/riscv64-unknown-elf-gcc: No such file or directory' I also tried doing `sudo /opt/riscv/riscv64-unknown-linux-gnu-gcc -V` but it said `sudo: /opt/riscv/riscv64-unknown-elf-gcc: command not found`. Can you help me with this? 
Thank you. I appreciate your help a lot.

Preetderasari14, 2021-02-01:
> Hi @OliverrrrWu thank you for your reply. I did exactly as you mentioned. I deleted the existing toolchain using `sudo rm -r /opt/riscv` and I think that worked. Then I did a fresh installation of [...]

I solved the issue. Now I am getting this as output when I run `riscv64-unknown-elf-gcc -v`:
`Using built-in specs.
COLLECT_GCC=riscv64-unknown-elf-gcc
COLLECT_LTO_WRAPPER=/opt/riscv/libexec/gcc/riscv64-unknown-elf/10.2.0/lto-wrapper
Target: riscv64-unknown-elf
Configured with: /home/preet/riscv-gnu-toolchain/riscv-gcc/configure --target=riscv64-unknown-elf --prefix=/opt/riscv --disable-shared --disable-threads --enable-languages=c,c++ --with-system-zlib --enable-tls --with-newlib --with-sysroot=/opt/riscv/riscv64-unknown-elf --with-native-system-header-dir=/include --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libgomp --disable-nls --disable-tm-clone-registry --src=.././riscv-gcc --enable-multilib --with-abi=lp64d --with-arch=rv64imafdc --with-tune=rocket 'CFLAGS_FOR_TARGET=-Os   -mcmodel=medlow' 'CXXFLAGS_FOR_TARGET=-Os   -mcmodel=medlow'
Thread model: single
Supported LTO compression algorithms: zlib
gcc version 10.2.0 (GCC) `

For those who face similar issue, I created a `~/.bash_aliases` file and added the line `export PATH=$PATH:/opt/riscv/bin`. After that I just rebooted my computer and then it worked.

Preetderasari14, 2021-02-01:
I am going to keep this issue open since the documentation of this repo didn't seem explanatory for someone inexperienced with gcc or even cross-compilers. So, if others have issues or want help from others they can add comments to this issue and people can help each other out. Also, I would still appreciate if anyone has compiled any binaries to use with the gem5 simulator and if they faced any issues with the default configuration of this toolchain.

Thanks and happy learning!

TommyMurphyTM1234, 2022-09-29:
PR contributions/enhancements to the main readme (https://github.com/riscv-collab/riscv-gnu-toolchain/blob/rvv-next/README.md) are welcome. But there is only so much that this can cover and it's beyond the scope of this README to explain to a complete beginner how to compile/cross compile the GCC toolchain. Also, these days there are several other mechanisms (e.g. crosstool-NG, xPack Project etc.) that can be used to compile/cross compile the tools, and there are also several options for downloading prebuilt binary versions of the tools obviating the need to build them from scratch.

closed 2022-09-29 by TommyMurphyTM1234