riscv-collab/riscv-gnu-toolchain #832: Error: illegal operands `addi $1,a5'

subhajit26, 2021-01-27:
Hi,
     I am trying this code  and compiling with riscv32-unknown-elf-gcc but it gives me the illegal operand error. I also tried with add but then then the error is same. My base instruction set is RV32IMC. Can you please help  me to solve the error and get an understanding of where am I going wrong.
int src = 1;
int dst;

asm ("mov %1, %0\n\t"
    "add $1, %0"
    : "=r" (dst)
    : "r" (src));

printf("%d\n", dst);

kito-cheng, 2021-01-27:
Seems like you need `%1` rather than `$1`.

subhajit26, 2021-01-27:
Hi,
       I am adding an imm value which is 1 and it is said in the GNU page that imm should be prefixed by $ sign. If I give %1 will it be the same functionality as before that is adding ´1´ to the dest? Please suggest . If possible can you please refer to some resources from where I can get a clear picture of inline assembly for riscv elf toolchain since I need it for adding custom instructions.

subhajit26, 2021-01-27:
Hi,
      I tried with add %1, %0 and this time it is 2 addition between 2 register, but still I got the error of illegal operands.

kito-cheng, 2021-01-27:
`add` instruction require 3 operands but you only give 2.

subhajit26, 2021-01-27:
Hi Kito,
    Yeah, I did not see that , my mistake. But can u please tell me why addi instruction is not working with immediate value.

kito-cheng, 2021-01-29:
@subhajit26 I guess you are learning the RISC-V assembly programming, and I know the ISA spec manual is not very detailed on this part, so share you a trick to I used, observing the output of gcc, for example you want to know how to add an immediate value, then you could write an equivalent C program and compile with GCC like bellow : 

```
$ cat x.c
int foo(int x){
    return x+100;
}
$ riscv64-unknown-elf-gcc x.c -o - -S -O -march=rv32i -mabi=ilp32
...
foo:
        addi    a0,a0,100
        ret
...
```

And then you will got the assembly code for add a register with immediate value :)

TommyMurphyTM1234, 2022-09-27:
Question about how to write (GCC inline) assembly code and not an issue with the RISC-V GCC toolchain itself.

closed 2022-09-27 by TommyMurphyTM1234