riscv-collab/riscv-gnu-toolchain #833: Few Doubts Regarding Toolchain

afzalj95-alt, 2021-01-29:
Hi All,

Is there any command which only compiles C Program not executes it in the Gnu toolchain? 
And Also What is the way to provide the stimulus as a hex file from the higher-level language like C to RISC V Core

Thanks

jim-wilson, 2021-02-02:
The gcc command will compile to an ELF binary.  To get a hex file, you can try objcopy, but SiFive recommends to use instead
https://github.com/sifive/elf2hex
because the verilog output file support in objcopy has a bug.

lazyparser, 2021-03-03:
@afzalj95-alt Hi is this question got answered?

TommyMurphyTM1234, 2022-09-27:
No update/activity for over a year and in any case, it seems to be a question about how to use the tools as opposed to an issue with the RISC-V GCC toolchain itself.
Closing.

closed 2022-09-27 by TommyMurphyTM1234