riscv-collab/riscv-gnu-toolchain #834: Is it possible to list the C code corresponding to the disassembly file line

rahulraveendran15-coder, 2021-01-29:
Hi,
Is there any way to get the C code or source code corresponding to the disassembly lines?(Aim is to identify the c line from the disassembly file)

I need to create a profiler depending on the timing and instructions

kito-cheng, 2021-02-02:
You need to compile your program with `-g` and then you could use objdump with `-d -S` to get that.

e.g.
```
$ cat hello.c
#include <stdio.h>
int main(){
    printf("Aloha\n");
    return 0;
}
$ riscv64-unknown-elf-gcc hello.c -g
$ riscv64-unknown-elf-objdump -d -S a.out
...
000000000001014e <main>:
#include <stdio.h>
int main(){
   1014e:       1141                    addi    sp,sp,-16
   10150:       e406                    sd      ra,8(sp)
   10152:       e022                    sd      s0,0(sp)
   10154:       0800                    addi    s0,sp,16
    printf("Aloha\n");
   10156:       67c9                    lui     a5,0x12
   10158:       6b078513                addi    a0,a5,1712 # 126b0 <__errno+0xc>
   1015c:       1c0000ef                jal     ra,1031c <puts>
    return 0;
   10160:       4781                    li      a5,0 
}
   10162:       853e                    mv      a0,a5
   10164:       60a2                    ld      ra,8(sp)
   10166:       6402                    ld      s0,0(sp)
   10168:       0141                    addi    sp,sp,16
   1016a:       8082                    ret
...

```

rahulraveendran15-coder, 2021-02-03:
Thank you very much for the help.
Closing the issue.

closed 2021-02-03 by rahulraveendran15-coder