riscv-collab/riscv-gnu-toolchain #836: can't link soft-float modules with double-float modules

qs132, 2021-02-02:
Hello!
         I wanted to compile coremark with the rv32i, but the compiler reported errors. I deleted everything about floating-point in coremark code.

heyman@heyman-virtual-machine:~/CoreMark/coremark$ riscv32-unknown-elf-gcc -march=rv32i -mabi=ilp32 -T riscv.ld -g  core_list_join.c core_main.c core_matrix.c core_state.c core_util.c core_portme.c -I ./  
/path/to/install/riscv/toolchain/lib/gcc/riscv32-unknown-elf/10.0.0/../../../../riscv32-unknown-elf/bin/ld: /tmp/ccxIQXE8.o: can't link soft-float modules with double-float modules
/path/to/install/riscv/toolchain/lib/gcc/riscv32-unknown-elf/10.0.0/../../../../riscv32-unknown-elf/bin/ld: failed to merge target specific data of file /tmp/ccxIQXE8.o
/path/to/install/riscv/toolchain/lib/gcc/riscv32-unknown-elf/10.0.0/../../../../riscv32-unknown-elf/bin/ld: /tmp/cciCVHNb.o: can't link soft-float modules with double-float modules
/path/to/install/riscv/toolchain/lib/gcc/riscv32-unknown-elf/10.0.0/../../../../riscv32-unknown-elf/bin/ld: failed to merge target specific data of file /tmp/cciCVHNb.o
/path/to/install/riscv/toolchain/lib/gcc/riscv32-unknown-elf/10.0.0/../../../../riscv32-unknown-elf/bin/ld: /tmp/ccxdXA1e.o: can't link soft-float modules with double-float modules
/path/to/install/riscv/toolchain/lib/gcc/riscv32-unknown-elf/10.0.0/../../../../riscv32-unknown-elf/bin/ld: failed to merge target specific data of file /tmp/ccxdXA1e.o
/path/to/install/riscv/toolchain/lib/gcc/riscv32-unknown-elf/10.0.0/../../../../riscv32-unknown-elf/bin/ld: /tmp/ccI6RLji.o: can't link soft-float modules with double-float modules
/path/to/install/riscv/toolchain/lib/gcc/riscv32-unknown-elf/10.0.0/../../../../riscv32-unknown-elf/bin/ld: failed to merge target specific data of file /tmp/ccI6RLji.o
/path/to/install/riscv/toolchain/lib/gcc/riscv32-unknown-elf/10.0.0/../../../../riscv32-unknown-elf/bin/ld: /tmp/ccjei2El.o: can't link soft-float modules with double-float modules
/path/to/install/riscv/toolchain/lib/gcc/riscv32-unknown-elf/10.0.0/../../../../riscv32-unknown-elf/bin/ld: failed to merge target specific data of file /tmp/ccjei2El.o
/path/to/install/riscv/toolchain/lib/gcc/riscv32-unknown-elf/10.0.0/../../../../riscv32-unknown-elf/bin/ld: /tmp/ccC2yt2o.o: can't link soft-float modules with double-float modules
/path/to/install/riscv/toolchain/lib/gcc/riscv32-unknown-elf/10.0.0/../../../../riscv32-unknown-elf/bin/ld: failed to merge target specific data of file /tmp/ccC2yt2o.o
collect2: error: ld returned 1 exit status

How can I fix it?
Thank you!

kito-cheng, 2021-02-02:
It's because your C library still using hard floating-point instruction, so I would suggest you re-build toolchain with `--with-arch=rv32i --with-abi=ilp32` or enable multi-lib via `--enable-multilib`

qs132, 2021-02-03:
> It's because your C library still using hard floating-point instruction, so I would suggest you re-build toolchain with `--with-arch=rv32i --with-abi=ilp32` or enable multi-lib via `--enable-multilib`

fix it!
Thank you very much!

closed 2021-02-03 by qs132

Yeting1029, 2022-05-20:
Try ../configure --prefix=$RISCV --host=riscv64-unknown-elf  --with-arch=rv64gc --with-abi=lp64