riscv-collab/riscv-gnu-toolchain #839: Problem with RV32IF gcc compiler

sudharsan2000, 2021-02-08:
Compiler built with arch = 'RV32IF' and abi = 'lp32f'.

A very simple test C program is taken. 
``` C
int main() {
        float a=1000.1, b=3.5;
        float c, d= 10;
        c = a + b;
        d-=c;
}
```

This program compiled with the built toolchain using:
    ` riscv32-unknown-elf-gcc -c -g prog.c -o output.o`
followed by:
`riscv32-unknown-elf-objdump -d output.o`

Gives the following output:
```
output.o:     file format elf32-littleriscv

Disassembly of section .text:

00000000 <main>:
   0:   fe010113                addi    sp,sp,-32
   4:   00812e23                sw      s0,28(sp)
   8:   02010413                addi    s0,sp,32
   c:   000007b7                lui     a5,0x0
  10:   0007a787                flw     fa5,0(a5) # 0 <main>
  14:   fef42627                fsw     fa5,-20(s0)
  18:   000007b7                lui     a5,0x0
  1c:   0007a787                flw     fa5,0(a5) # 0 <main>
  20:   fef42427                fsw     fa5,-24(s0)
  24:   000007b7                lui     a5,0x0
  28:   0007a787                flw     fa5,0(a5) # 0 <main>
  2c:   fef42227                fsw     fa5,-28(s0)
  30:   fec42707                flw     fa4,-20(s0)
  34:   fe842787                flw     fa5,-24(s0)
  38:   00f777d3                fadd.s  fa5,fa4,fa5
  3c:   fef42027                fsw     fa5,-32(s0)
  40:   fe442707                flw     fa4,-28(s0)
  44:   fe042787                flw     fa5,-32(s0)
  48:   08f777d3                fsub.s  fa5,fa4,fa5
  4c:   fef42227                fsw     fa5,-28(s0)
  50:   00000793                li      a5,0
  54:   00078513                mv      a0,a5
  58:   01c12403                lw      s0,28(sp)
  5c:   02010113                addi    sp,sp,32
  60:   00008067                ret
```

The issue is the way Floating Point register's data are loaded.
According to the C program the initial floating point values- 1000.1, 3.5 and 10 have to be loaded from the Program Memory, but instead they are loaded using the FLW instructions which use the Data Memory(whose contents won't be of significance).

The data should be loaded by a LUI command to an integer register, a store instruction and then a FLW command right? 

Am I missing something?

P.S No errors were reported while building the toolchain or while compiling the program.

jim-wilson, 2021-02-08:
The compiler will put FP constants in the data section.  That is normal practice.  If you can't load from data then a lot of programs aren't going to work as a lot of stuff goes in the data section.  A normal program loader should load the data section into memory along with the text section.

sudharsan2000, 2021-02-09:
Ok. Thanks.
But the addresses for loading from the data section are all the same, is that normal?

jim-wilson, 2021-02-09:
You are looking at an unrelocated object file.  Use objdump -dr to see the relocations.  When you link, the relocations will be replaced with actual addresses.

sudharsan2000, 2021-02-09:
Thanks a lot for helping me out!!

closed 2021-02-09 by sudharsan2000