riscv-collab/riscv-gnu-toolchain #841: Linking soft-float versions of sin/cos functions in math.h

nagendragd, 2021-02-13:
The below is a question rather than an issue:
I have set up my riscv gnu tool chain for rv32gc.
My code has calls to sin/cos functions declared in math.h

I am compiling with -march=rv32imac -mabi=ilp32:

Linking the code with -lm -L<path to libm.a>, but yet the linker is not able to find sin(.).
I get this linker error:
compute.c:(.text+0x310): undefined reference to `sin'

I have the same problem with being able to link in the 'pow' function too.

Any pointers are appreciated.  

Best Regards
Nagendra

jim-wilson, 2021-02-13:
You didn't include enough info to reproduce, such as the full command line being used.  I would guess that -lm is not at the end of the gcc command.  It must come after the file that calls sin.

You mentioned that the toolchain was configured for rv32gc but not the ABI.  If you are using a -mabi option different than the configuration, then this can only work if the toolchain was configured multilib, and you are using a arch/abi configuration that is one of the multilibs.  Otherwise, precompiled libraries like the math library will only be compiled for the default arch/abi which are not link compatible with your just compiled code.

nagendragd, 2021-02-13:
Jim,

Thank you. You solved it for me. I forgot about the good old “put the
dependency libs at the end” rule.
I moved the -lm to the end and it worked!!

Thank you!

And thanks also for the other notes.

Best regards
Nagendra

nagendragd, 2021-02-13:
Marking as closed

closed 2021-02-13 by nagendragd