riscv-collab/riscv-gnu-toolchain #843: Assembler in riscv gnu toolchain

pbhatt173, 2021-02-17:
Hi,

Is assembler available in riscv gnu toolchain ?

stnolting, 2021-02-17:
What do you mean exactly?
`*-as` is the *assembler executable* as part of the toolchain. For example:

```
./riscv32-unknown-elf-as -version
GNU assembler (GNU Binutils) 2.34
Copyright (C) 2020 Free Software Foundation, Inc.
This program is free software; you may redistribute it under the terms of
the GNU General Public License version 3 or later.
This program has absolutely no warranty.
This assembler was configured for a target of `riscv32-unknown-elf'.
```

pbhatt173, 2021-02-18:
Hi,

Can we use GNU assembler as reference model like ISS ?

jim-wilson, 2021-02-18:
Not clear what you are asking here.  The assembler is just an assembler.  It isn't a reference model.  Are you asking about the psABI perhaps?  That is in riscv-elf-psabi-doc.  GNU as does follow the psabi.

pbhatt173, 2021-02-18:
Hi,

In core-v-verif repository we have ISS simulator tool, which is used as a reference model in uvm verification environment.
So my doubt is can we use these GNU assembler as a ISS simulator as reference model

LevyHsu, 2021-02-18:
The assembler is used to "translate" assembly code to machine code, it does not perform the function of an ISS (Some of ISS can read assembly code directly, which means it got an builtin assembler) 
Try qemu or spike maybe?

pbhatt173, 2021-02-18:
okay..... thanks

closed 2021-03-03 by lazyparser