riscv-collab/riscv-gnu-toolchain #846: Position independent code support with ELF newlib flavor

agrobman, 2021-03-02:
Does the toolchain support position independent code compilation/executables with newlib flavor and ELF output?
If so then how ( any doc reference to how to build loader for such executables )

jim-wilson, 2021-03-05:
There is no RISC-V newlib support for shared libraries, so the RISC-V embedded ELF linker does not support shared libraries.  This is trivial to re-enable if you want to experiment.  In binutils, in ld/emulparams/elf32lriscv-defs.sh, look at the GENERATE_SHLIB_SCRIPT and GENERATE_PIE_SCRIPT which are only set for non-embeddedl-elf targets.

There is no RISC-V embedded PIC ABI defined, so the compiler doesn't have any special embedded elf pic support.  It will let you generate PIC, but it probably won't run without a special loader, which you probably have to write yourself.

I would suggest looking at boot loaders.  Some of them compile themselves pic, and have special support to load their pic code into memory to run it.  I don't know if any have such support for RISC-V though.

TommyMurphyTM1234, 2022-11-10:
Closed for the same reason outlined here:

https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1039#issuecomment-1309095288