riscv-collab/riscv-gnu-toolchain #848: Can we add a few tags to triage bugs in issue lists and separate real bugs, questions, and feature requests?

lazyparser, 2021-03-03:
Hi maintainers,

There are 150+ open issues now. It is not convenient for new contributors to pick a first easy bug to fix.

Can we improve it? e.g. add a 'good-first-bug' tag and tag the bugs?

My colleagues and I would like to contribute on the bug triage and issue management. Feel free to ping me if you think it is helpful. :)

kito-cheng, 2021-03-03:
That's sounds good to me, I can grant you permission to doing this.

kito-cheng, 2021-03-03:
@lazyparser you should able to add label for issues now.

lazyparser, 2021-03-03:
> @lazyparser you should able to add label for issues now.

Thanks!

closed 2021-03-03 by lazyparser

jim-wilson, 2021-03-04:
Very few of these riscv-gnu-toolchain issues are interesting.  Most of them are newby questions that have already been answered.  You will find higher quaitly things to fix in the GNU binutils/gcc/etc bugzillas.  I used to go through and close them, but I haven't had time for that in a while.

Don't forget that we also have riscv-binutils-gdb, riscv-gcc, riscv-newlib, riscv-glibc, and riscv-dejagnu.  All of those have open issues also.  However, we would like to kill off all of the submodules and just use FSF/sourceware trees directlly, so long term you should not encourage anyone to ask questions in those submodules.  We do want to keep riscv-gnu-toolchain as a newby build script, so it is still OK to ask questions by raising issues here.

There is also riscv-tools.  It used to contain a broken toolchain (gcc-7.2), now it contains no toolchain.  But some people still ask toolchain questions there.  riscv-tools hasn't been properly maintained for about 2 years.  If you look at the submodules, most of them about are 2 years out of date.  If someone wants to volunteer to fix that, that would be useful.  riscv-tools is mostly stuff useful for hardware engineers, but it has a simulator spike that may be useful to software folk.  qemu is faster due to the jit, but spike is easier to modify, so new ISA extensions are likely to appear in spike before qemu.

lazyparser, 2021-03-05:
> Very few of these riscv-gnu-toolchain issues are interesting. Most of them are newby questions that have already been answered. You will find higher quaitly things to fix in the GNU binutils/gcc/etc bugzillas. I used to go through and close them, but I haven't had time for that in a while.

My toolchain colleagues are working on the upstream bugzillas. I will go through and closed the answered issues every week.

> Don't forget that we also have riscv-binutils-gdb, riscv-gcc, riscv-newlib, riscv-glibc, and riscv-dejagnu. All of those have open issues also. However, we would like to kill off all of the [...]

Sure. I want to reduce the number open issues (maintain it as a good testing process). After I walk through the meta repo, I will dig into submodules.

> There is also riscv-tools. It used to contain a broken toolchain (gcc-7.2), now it contains no toolchain. But some people still ask toolchain questions there. riscv-tools hasn't been properly [...]

Sure. The PLCT Lab is glad to maintain it. We may start from building enough CI for riscv-tools, and then bump it to up to date.