riscv-collab/riscv-gnu-toolchain #850: The CI for macOS build is failing (before 407cdc0, 2021-03-01)

lazyparser, 2021-03-04:
build failed on macOS after 407cdc0. I checked it, figured out that it was a network/internal issue of Travis-CI, not the riscv-gnu-toolchain repo.

Do we have any process / testing strategy for dealing with these alarms?

lazyparser, 2021-03-04:
https://travis-ci.org/github/riscv/riscv-gnu-toolchain/jobs/760431847

kito-cheng, 2021-03-04:
Honestly I am almost give up the `travis-ci ` now, I guess maybe we should migrate to github action.

ilg-ul, 2021-03-04:
I don't know how Travis works for paid accounts, but for open source projects Travis became almost useless; I already migrated all my CI jobs to GitHub Actions; so far, everything is fine. Recommended.

lazyparser, 2021-03-04:
> Honestly I am almost give up the `travis-ci` now, I guess maybe we should migrate to github action.

Then let's remove the false alarm by removing the travis-ci (especially on the README.md, which may confuses people.)

> I don't know how Travis works for paid accounts, but for open source projects Travis became almost useless; I already migrated all my CI jobs to GitHub Actions; so far, everything is fine. Recommended.

+1

jim-wilson, 2021-03-04:
I gave up on travis a while ago.  I think we have a free account, and we can't get enough disk/cpu to successfully run toolchain regressions tests.  I do my own tests by hand when I need them.  And I ignore all travis failures as uninteresting when I have my own test runs that succeeded.

I didn't know we have MacOS host testing via travis.  That would be useful since I don't have a mac, but only if it was something other than travis that actually worked.

ilg-ul, 2021-03-04:
> I didn't know we have MacOS host testing via travis.

There are Intel macOS VMs on GitHub Actions too.

lazyparser, 2021-03-05:
Now seems

1. we are all OK to use other CI infrastructure instead of Travis-CI. I am glad to volunteer if needed.
2. MacOS build is optional. Better to have one, but should not block any commits. That said, as a tier-3 platform.

The goal of this issue is:
1. To remove the false failure notice on README.md and commit status.
2. Add more build configs and regression tests.

So, how to move forward?

kito-cheng, 2021-03-05:
There is already some github action added by @kevinpark1217 before https://github.com/riscv/riscv-gnu-toolchain/tree/master/.github 

So I think the work item is:
1. Turn off the Travis-ci and remove false failure notice on README.md 
2. Add regression logic into github action for Ubuntu only.
3. Add PR checker for github action.
4. Add more build configs and regression tests, e.g. MacOS or windows build ...

closed 2021-03-05 by lazyparser