riscv-collab/riscv-gnu-toolchain #852: [Build Failure] riscv-gnu-rv64gc-master/qemu/linux-user/ioctls.h:225:9: error: ‘SIOCGSTAMP’ undeclared here (not in a function); did you mean ‘SIOCSRARP’?

lazyparser, 2021-03-05:
Build Script:

```
# 407cdc0ceb0f6d760342ac6cf1b890378f5a81bd
git clone https://github.com/riscv/riscv-gnu-toolchain
cd riscv-gnu-toolchain

# master branch

RV64_BIN="$PWD/obj-rv64"

./configure --prefix="$RV64_BIN"
make linux -j $(nproc)

make report-linux  -j $(nproc) SIM=qemu # Run with qemu
```

https://ci.rvperf.org/job/riscv-gnu-rv64gc-master/26/console

```
/home/jenkins_node/workspace/riscv-gnu-rv64gc-master/qemu/linux-user/ioctls.h:225:9: error: ‘SIOCGSTAMP’ undeclared here (not in a function); did you mean ‘SIOCSRARP’?
  225 |   IOCTL(SIOCGSTAMP, IOC_R, MK_PTR(MK_STRUCT(STRUCT_timeval)))
      |         ^~~~~~~~~~
/home/jenkins_node/workspace/riscv-gnu-rv64gc-master/qemu/linux-user/syscall.c:4913:23: note: in definition of macro ‘IOCTL’
 4913 |     { TARGET_ ## cmd, cmd, #cmd, access, 0, {  __VA_ARGS__ } },
      |                       ^~~
```

seems QEMU bug.

lazyparser, 2021-03-05:
OS: Ubuntu 20.04

lazyparser, 2021-03-05:
> seems QEMU bug.

I think this is not a QEMU bug, maybe update submodule would help. Not sure.

jim-wilson, 2021-03-05:
Looks like this problem
    https://patchwork.kernel.org/project/qemu-devel/patch/20190617114005.24603-1-berrange@redhat.com/
And this fix isn't in our tree.  This is due to linux kernel changes, so will only show up on a recent distro.

kito-cheng, 2021-03-05:
I believe bump qemu to 5.2.0 could fix that, I was working on bump glibc, qemu and linux kernel last month, but got few extra failed case on rv32gc-linux, I didn't have time to figure it out what happen so I didn't bump that yet, maybe I can create a branch to let you try.

lazyparser, 2021-03-05:
> Looks like this problem
> https://patchwork.kernel.org/project/qemu-devel/patch/20190617114005.24603-1-berrange@redhat.com/
> And this fix isn't in our tree. This is due to linux kernel changes, so will only show up on a recent distro.

Thanks. Seems like multiple OS testbed are helpful :)
I'd add more test platform later.

> I believe bump qemu to 5.2.0 could fix that, I was working on bump glibc, qemu and linux kernel last month, but got few extra failed case on rv32gc-linux, I didn't have time to figure it out what happen so I didn't bump that yet, maybe I can create a branch to let you try.

No need to create a new branch for this. I will modify the build script temporarily to use 5.2.0. There are already too many branches in the tree :)

( I'd also ask my QEMU colleagues take a look. @GeassCore @liweiwei90 )

lazyparser, 2021-03-07:
In Ubuntu 20.04, bump to qemu v5.2.0 solves the issue. Now I can run all regression tests.

A few potential bugs were observed:

https://ci.rvperf.org/blue/organizations/jenkins/riscv-gnu-toolchain-master-detailed-regresstion-2004/detail/riscv-gnu-toolchain-master-detailed-regresstion-2004/9/pipeline/47

And it literally takes more than 12 hours for running all the ~30 configs. In order to not bother watchers of the repo, these failures will be filed in PLCT's downstream repo before Jiawei figure out they are real GCC bugs or merely CI issues.

I've also built a regression testing job for riscv-gnu-toolchain in Ubuntu 18.04:

https://ci.rvperf.org/blue/organizations/jenkins/riscv-gnu-toolchain-master-detailed-regresstion-ubuntu1804/detail/riscv-gnu-toolchain-master-detailed-regresstion-ubuntu1804/3/pipeline

No failures/bugs observed yet.

@kito-cheng Hi, based on the two CI jobs above (as baseline), we can now try to bump different combinations automatically. Do you have some suggestions or practical bump heuristics that I can follow?

As the first try, I simply ran `git submodule update -f --init --recursive --remote`, and things were not going well. Quite a few failures. Not sure whether worth to dig into it, I just stopped at the first try.

kito-cheng, 2021-03-08:
The reason why I still not bumping qemu is rv32 linux support, it got extra fails due to ABI changes, so we need to bump glibc and kernel header at same time to fix the issue.

But I got another fails after bumping glibc, kernel header and qemu...so it never get bumped.

closed 2021-03-08 by lazyparser