riscv-collab/riscv-gnu-toolchain #853: work through the open issues and close all issues seems answered/solved.

lazyparser, 2021-03-05:
> > Since these aren't paying customers, I would suggest that if we think a problem has been answered, and it has been more than a few weeks since the original poster responded, that we should just close it. The original poster can object if they think it isn't fixed yet. I think this approach is less work for us, since few are likely to object to closing issues for answered questions.
> 
> Fair enough. I will work through the open issues and close all issues seems answered/solved. If I closed it wrong, anyone can just reopen it.

kito-cheng, 2021-03-05:
@lazyparser Thanks, I am really appropriate you jump in this work! I believe maintain this repo is meaningful for RISC-V community so I maintain this for long time, that's really great to having you to make here better.

TommyMurphyTM1234, 2022-09-27:
I am doing this right now (reviewing the open issues and first of all closing any that are obviously redundant) so that makes this issue redundant. 🙂