riscv-collab/riscv-gnu-toolchain #855: Add regression logic into github action for Ubuntu only.

lazyparser, 2021-03-05:
> There is already some github action added by @kevinpark1217 before https://github.com/riscv/riscv-gnu-toolchain/tree/master/.github 

So I think the work item is:
1. Turn off the Travis-ci and remove false failure notice on README.md 
2. Add regression logic into github action for Ubuntu only.
3. Add PR checker for github action.
4. Add more build configs and regression tests, e.g. MacOS or windows build ...

_Originally posted by @kito-cheng in https://github.com/riscv/riscv-gnu-toolchain/issues/850#issuecomment-791195941_

jim-wilson, 2021-03-05:
I don't know if we want to use github actions for a full regression build.  That uses a lot of CPU power.  Maybe start with something smaller, like a few select configs, maybe rv32imac/ilp32/elf and rv64gc/lp64d/linux and then if it works well we can try doing more.  Otherwise, it sounds like a reasonable plan.

lazyparser, 2021-03-07:
Indeed. The full regression testing takes 8~12 hours on a buildbot with 16 core Xeon and 128G RAM.

Some practical ideas:
1. separate the `rv32imac/ilp32/elf` and `rv64gc/lp64d/linux` from the full regression tests as Jim suggested. this CI is intended to block the PR if fails.
2. Run other configs in the full regression tests as a separate CI job, which does not block the reviewers approve & merge the PR.
3. Build Can be triggered by webhooks, which means @kito-cheng (or other owners/mantainers of this repo) can run these time consuming jobs on ci.rvperf.org, given that github only provides a 2 core build machine.
4. We can also have post-full-regression-test, which is already in place.

TommyMurphyTM1234, 2022-10-15:
@cmuellner or @kito-cheng - do either of you know if this issue is still extant or can be closed?

cmuellner, 2022-11-16:
I cannot see any build failures or broken build indicators.
Closing in doubt of any pending action items.
Feel free to reopen if I missed something.

closed 2022-11-16 by cmuellner