riscv-collab/riscv-gnu-toolchain #856: Add PR checker for/using github action.

lazyparser, 2021-03-05:
There is already some github action added by @kevinpark1217 before https://github.com/riscv/riscv-gnu-toolchain/tree/master/.github 

So I think the work item is:
1. Turn off the Travis-ci and remove false failure notice on README.md 
2. Add regression logic into github action for Ubuntu only.
3. Add PR checker for github action.
4. Add more build configs and regression tests, e.g. MacOS or windows build ...

_Originally posted by @kito-cheng in https://github.com/riscv/riscv-gnu-toolchain/issues/850#issuecomment-791195941_

TommyMurphyTM1234, 2022-10-15:
@cmuellner or @kito-cheng - do either of you know if this issue is still extant or can be closed?

cmuellner, 2022-11-16:
This is a duplicate of #855.

closed 2022-11-16 by cmuellner