riscv-collab/riscv-gnu-toolchain #858: [win32] Add more build configs and regression tests: windows build

lazyparser, 2021-03-05:
There is already some github action added by @kevinpark1217 before https://github.com/riscv/riscv-gnu-toolchain/tree/master/.github 

So I think the work item is:
1. Turn off the Travis-ci and remove false failure notice on README.md 
2. Add regression logic into github action for Ubuntu only.
3. Add PR checker for github action.
4. Add more build configs and regression tests, e.g. MacOS or windows build ...

_Originally posted by @kito-cheng in https://github.com/riscv/riscv-gnu-toolchain/issues/850#issuecomment-791195941_

aswaterman, 2021-03-05:
@lazyparser you are creating an enormous amount of traffic on this list.  I am unfollowing this repo.

kito-cheng, 2021-03-05:
@lazyparser maybe try https://github.com/riscv/riscv-gnu-toolchain/projects ? I didn't use that before, but it seems might be useful.

lazyparser, 2021-03-05:
> @lazyparser you are creating an enormous amount of traffic on this list. I am unfollowing this repo.

Hi @aswaterman 

Oh, Sorry. Please keep tracking this repo, because there are quite a few users asking here.

My actions on this repo would be no more than ~35 per week hereafter. i promise :-)

> @lazyparser maybe try https://github.com/riscv/riscv-gnu-toolchain/projects ? I didn't use that before, but it seems might be useful.

Good idea. I will try to use the projects. Thank you.

TommyMurphyTM1234, 2022-10-15:
@cmuellner or @kito-cheng - do either of you know if this issue is still extant or can be closed?

cmuellner, 2022-11-16:
Similar like #857.

The CI seems to work for quite some time already.
So we can close this issue.

Current builders are only building on Ubuntu (esp. no Windows and no MacOS), but given that this is the main development target, this is ok for me.

closed 2022-11-16 by cmuellner